marcocorvi/topodroid #8: translations fixes

marcocorvi, 2016-02-26:
a code analysis with lint showed that texts had several issues

- **missing translations**
  for several strings this may be a false issue as the string does not need to be translated
  (examples: acronym like "DXF", the *Value string arrays, the name of the help-pages)

- **typography**

- **typos **

- **plurals**

- **hardcoded** text (these are special chars and are ok)

- **unused strings**
  there are a few unused strings that are not commented

- **commented strings**
  many strings in the string files are commented UNUSED
  a revision is useful in order to drop string that are not referenced even in 
  commented code lines (string referenced in comments must be kept in)

ThiloM, 2016-02-27:
Hi Marco,
I translatet the german issues. I found only one file with "TODO"s.
Cheers
Thilo

<http://www.arge-grabenstetten.de/www2/wir-uber-uns/newsletter/>

fato63, 2016-02-27:
Ciao Marco,
ho rivisto il file strings.xml togliendo i "TODO" nuovi ed i "DONE" che
avevo lasciato la volta scorsa.
Ho visto che molte stringhe "UNUSED" sono tradotte, penso sia un tuo
intervento...
si tratta di traduzioni non esatte o per le quali e' piu' utile mantenere
il termine in inglese?
Attendo tue nuove e qualche indicazione per fare meglio le prossime!
Cordiali saluti, Fabrizio

marcocorvi, 2016-02-28:
thanks you all for the review of translations.

the "marks":

TODO is a string that needs be translated,
usually a new string that i added to the file

UNUSED is an old translated string that is no longer used, but i rather
leave in the file either because it is commented in the source or because i
dropped, but i am not sure i might have to put it back.
(now and then i try to remove strings that have been unused for long time)

OK is a string that does non need be translated, eg, the name of a program
("Compass"). lint gives a warning about these: do not take it into account.

also the string arrays for *Value must not be translated, and may not
appear in the translation files.
the array of languages is special, as it needs no translarion.

FIXME is a translated string for which i have updated the english version.
it is a hint to check the translation with the updated string.

marco

marcocorvi, 2016-03-05:
plurals:

there are a few strings that depends on a "quantity".
these are in plurals.xml

how much you need to specialize the string depends of your language.

in some languages it is not necessary to specialize the string by the quantity. in this case the item with qyantity "other" is all you need to add.

other languages differentiate zero from a positive number. in this case you need to add an item for "zero" and one for "other".

other languages are more detailed and you need also "one" and maybe "two"

marco

guillaumechardin, 2016-03-10:
Hi Marco,
i'm actually working on a big release of french translation.

How to tell some people on gihub to not edit fr files ? I have to start a
new branch to be sure no one will update those files until i release them ?

Does all modifications you add to master files will be added later on my
release ?

Bye.

Guillaume.

marcocorvi, 2016-03-10:
no worry.

basically you are the only person editing french files.
and if anyone else does it, git takes care of merging or signalling
conflicts (that it cannot resolve, but they are usually trivial for the
translations).

i might add in strings. but if you do not reorder them, ie, you keep them
in the same order as in the english file, git will nicely merge my
additions with your changes.

if you want to compile and test the translation before it is merged in the
master branch, you can start a new branch.
otherwise, there is no point in that.

marco

FranckGaragalh, 2016-03-11:
Bonjour Guillaume ;) j avais pris ta suite sur les dernières traductions
car le français était voué à disparaître . si tu reprends la suite c est
super car je n ai pas eu le temps de m y remettre;) promis je touche à
rien. Si tu as besoin d'aide par contre pas de soucis

@+ franck
Hi Marco,
i'm actually working on a big release of french translation.

How to tell some people on gihub to not edit fr files ? I have to start a
new branch to be sure no one will update those files until i release them ?

Does all modifications you add to master files will be added later on my
release ?

Bye.

Guillaume.

guillaumechardin, 2016-03-11:
Bonjour Franck !

je vais effectivement avoir besoin d'un peu d'aide !! Il y a des nouveaux
termes que je ne comprend pas !

Voici la liste des termes que je n'ai pas encore traduits. J'ai quelques
hésitations sur l'utilisation du terme splay(s). A savoir si cela indique
les visées latérales ou simplement certaines lignes de dessins, il faut
donc vérifier le contexte.

    Line 512:     
    Line 617:     
    Line 689:   
    Line 690:   
    Line 691:   
    Line 694:   
    Line 745:   
    Line 746:   
    Line 747:   
    Line 763:   
    Line 764:   
    Line 767:   
    Line 768:   
    Line 806:   
    Line 807:   
    Line 808:   
    Line 885:   
    Line 886:   
    Line 887:   
    Line 888:   
    Line 889:   
    Line 890:   
    Line 891:   
    Line 892:   
    Line 893:   
    Line 894:   
    Line 924:   
    Line 926:   
    Line 927:   
    Line 928:   
    Line 929:   
    Line 931:   
    Line 932:   
    Line 936:   
    Line 937:   
    Line 938:   

Je peux t'envoyer également ma version de fichier actuelle afin que tu
puisse la consulter et la relire avant publication.

Guillaume

marcocorvi, 2016-03-11:
hi
i think i should try to explain some of the strings, because the string
file has no context info.
hoping this helps (at least to find the point in the program where the
features are used).

splays are the lateral/trasversal shots taken from a station. the endpoint
of the splay is left unnamed.

prefix sketch_ refer to 3d sketching, an experimental feature much
undeveloped
and disabled.

prefix wall_ refers to autowall, an experimental feature to make topodroid
draw the walls around a leg using the splays.

azimuth refers to the setting of the reference azimuth which is used to
assign "extend" direction (left, vert, right) to the legs. this defines how
to draw the leg inthe extended ("developed") profile.

there are many options that defines how to handle splays in the sketches
(line style/"extend" direction), in the exports, and in the import (lrud
are converted to splays).
most of these are clino threshold and relative azimuth threshold.

backsight in the option of backsight fields in the form to manually enter a
measure.

timer_ has to do with using the android sensors to get an approx direction
(manual measure and ref azimuth). there is delay and beep volume.

side_drag is the ability to shift a sketch even in drawing mode, by
dragging it from the side. (when you use a stylus).

unit grid: sketch grid units (m or yd)

recent_ the most-recent drawing tool picker

area_border ??? probably the default visibility for therion export

cont_join has to do with joining the line you are drawing to an existing
one (must select the button, and start close to its end)

marco

fato63, 2016-03-17:
Ciao Marco,
ho terminato l'allineamento dei file strings.xml tra quello EN e quello IT.
Ora, a meno di qualche riga vuota, mi sembra ci sia tutto.
Saluti, Fabrizio

marcocorvi, 2016-03-17:
grazie.
ciao, marco

guillaumechardin, 2016-03-27:
Hi Marco,

here some few more questions for my current work, mostly is about more
explanations about what feature is to have the best translation in french.
Thanks for your assistance.

furthermore, can you tell me the email address of this group ? I always
replay to  "reply+[some msg UID]@reply.github.com ; and so this email in in
a "bad" thread/object !

Bye,
Guillaume

<string name="pref_leg_shots_title">Nombre minimum de tirs</string>
<string name="pref_leg_shots_summary">Nombre minimum de tirs par
visées</string>
is it the numbers of repeted shots to set/detect a shot as a leg ?

Backsight is a reverse shot ? Like this sequence (e-d is a
reverse/backsight) :
A-b b-c c-d e-d d-f

<string name="calibration_title">Calibration Instrument</string>
"instrument" is an external tool use to calibrate ?

can you add more info about that, is this a theshold to detect splay shots ?

Can you define me this one (i do not understand "splays station" - for me
either a splay OR a station...)

What is a dashed splay clino, is is just a visual feature  ? Set some splay
as dashed on sketch ? Or is it more than a visual preference ?

pref_shot_timer_title
this is while using android device compass ?

what is clino threshold for splay ? Is it a value to display splays as
dashed lines ? Is it only a display feature ? Or it is used internaly by
topodroid ?

*
where to find this one ?

Can you explain more ?

Same

Does thoses lines are still used ?

marcocorvi, 2016-03-27:
> Hi Marco,
>
> here some few more questions for my current work, mostly is about more
> explanations about what feature is to have the best translation in french.
> Thanks for your assistance.
>
> furthermore, [...]

there is no group.
i guess the mail is automatically send by github when someone post on a
issue,
and the message id is set by github.

> <string name="pref_leg_shots_title">Nombre minimum de tirs</string>
> <string name="pref_leg_shots_summary">Nombre minimum de tirs par
> visées</string>
> is it the numbers of repeted shots to set/detect a shot as a leg ?

yes, the minimum number.
can choose between two, three and four.
three is very convenient with distox2 firmware 2.4

in the dialog to enter a shot by hand you can also have fields for the
backsight, besides the foresight.
this is for when you do double reading:
forward and reverse.

> Backsight is a reverse shot ? Like this sequence (e-d is a
> reverse/backsight) :
> A-b b-c c-d e-d d-f
>
>
> <string name="calibration_title">Calibration Instrument</string>
> "instrument" is an external tool use to calibrate ?

again, manual shot data:
these are instruments calibration, eg 1m for a tape one meter short (start
at tag 1m).
this calibration is added to the value you enter in the dialog

to export to a program that does not have splay, topodroid computes lrud
using the splay that make at least this minimum angle with the shot

> 
> can you add more info about that, is this a theshold to detect splay
shots ?

see above

>
> 
> Can you define me this one (i do not understand "splays station" - for me
> either a splay OR a station...)

survex export.
in survex splays can be without to-station.
with this option the to station is set automatically.
for example splays at station 1 have to station 1a 1b etc.

>
> 
> What is a dashed splay clino, is is just a visual feature ? Set some splay
> as dashed on sketch ? Or is it more than a visual preference ?

just a display feature.

> pref_shot_timer_title
> this is while using android device compass ?

yes.
there is a delay, before it starts measuring.

> 
> what is clino threshold for splay ? Is it a value to display splays as
> dashed lines ? Is it only a display feature ? Or it is used internaly by
> topodroid ?

display only.
prefs can have: title summary message

> *l
> where to find this one ?

3d "sketching":
orient the 3d sketch using android sensors.

probably, no longer used
>
> 
> Can you explain more ?

autowall.
minimum distance between two splay point. to be both used.

>
> 
> 
> 
> Same

this is for autowalls.
says how far apart should wall point be.

> Does thoses lines are still used ?

yes.
you do not need to translate them.

marco

>
> > Ciao Marco,
> > ho terminato l'allineamento dei file strings.xml tra quello EN e quello
IT.

guillaumechardin, 2016-03-30:
Hi Marco !

Can you give more information about that :
<string name="sketch_no_split">No refinement. Decrease side max. length to
refine further.</string>

This one last line and fr translation is done  ! Still need re-readers !

FranckGaragalh, 2016-03-30:
Bonjour Guillaume . envoie moi les relectures à franckm@gmail.com ;)

guillaumechardin, 2016-04-06:
Bonjour Franck,

Je n'avais pas vu ton message, désolé !

Voici le fichier en PJ. Une relecture complète est à réaliser car j'ai
repris beaucoup d'éléments,
J'ai de grosses hésitations sur les termes concernant les chaines :
pref_walls_*. Je n'ai pas réussi précisément à transcrire dans un francais
intelligible la signification de ces éléments notament :
pref_walls_xclose_* et pref_walls_xstep_*
J'ai fait un peu comme j'ai pu :)

J'ai également laissé dans le fil des traduction LRUD. Qu'en penses tu ? On
francise ?

closed 2016-05-23 by marcocorvi