marcocorvi/topodroid #16: ukrainian and other translations

marcocorvi, 2016-05-23:
This is an "abriged" output of lint.

Please review the translation warnings that refer to the language you are maintaining.

lint is far from perfect,
therefore some warning do not need to be taken into account
( should make an exception file, but i'm lazy/busy ).
ignore those.

Tapter, 2016-05-25:
Could you attach the output; or where do i see that?

marcocorvi, 2016-05-25:
oops.
here is the link

[lint.txt](https://github.com/marcocorvi/topodroid/files/281496/lint.txt)

marcocorvi, 2016-06-07:
dropped old lint outputs from int18
and added a new one.
@all please check (possible) issues with your translation
thanks

marcocorvi, 2016-10-11:
hungarian translation

i noticed that hungarian translation contains english strings as well as the translations.
it is likely that this makes the strings too long so that, at times, they exceed the widgets space,
with the result that the translated string is cut at the first few chars.

what is the motivation for this choice ?
would not be better to keep only the translation ?

Szifon, 2016-10-11:
Sorry for any disturbances. I have no routine in the use of GIT and in
coding. I keep the English string temporary because I would like to recheck
the translation when I have finished it. I will delete them.

Péter Sűrű

marcocorvi, 2016-10-11:
that's ok

i was just wondering why the double strings.
i appreciate your help, and that of all others, for the translations,
And the comments to make the app better.

marco

fato63, 2016-12-20:
Ciao Marco,
I have just updated the IT String.xml file.
I have found, about at line 883, a group of lines in English without "TODO".
Is it correct?

Fabrizio

marcocorvi, 2016-12-20:
Sorry, i must have missed the TODOs

if a string has an english value it is a TODO.

marco

closed 2018-06-07 by marcocorvi