marcocorvi/topodroid #22: SVG output of points issue

caveradam, 2016-10-23:
Using Version 3.1.6i.

All points are placed at the origin of the viewbox, instead of at the point where they were actually placed. They show up as a blob and need to be replaced. 

Full file:
[image]

Zoomed into blob of point items:
[image]

Re-arranged to show the points that were in the blob:
[image]

marcocorvi, 2016-10-24:
could not reproduce this:
[1] points in the svg file have proper "translate"
[2] embedded in html the svg file displays points correctly in a browser

marcocorvi, 2019-10-24:
i could finally reproduce the issue with inkscape.
fixed in 4.1.4D

closed 2019-10-24 by marcocorvi