marcocorvi/topodroid #26: Cannot resolve symbols

DSBork, 2017-04-13:
Hi,

I cannot compile a project due to "Cannot resolve symbol" error in reloadSymbols function.

These lines cannot be compiled:
if ( mine   ) installSymbols( R.raw.symbols_mine,   true );
if ( geo    ) installSymbols( R.raw.symbols_geo,    true );
if ( archeo ) installSymbols( R.raw.symbols_archeo, true );
if ( paleo  ) installSymbols( R.raw.symbols_paleo,  true );
if ( bio    ) installSymbols( R.raw.symbols_bio,    true );

Can you please add to git these files:
./res/raw/symbols_mine
./res/raw/symbols_geo
./res/raw/symbols_archeo
./res/raw/symbols_paleo
./res/raw/symbols_bio

Or can you please propose other solution but commenting this code out.

marcocorvi, 2017-04-13:
you should shell make_symbols

closed 2017-04-13 by marcocorvi

DSBork, 2017-04-17:
Thank you for your answer, it almost works.

There are lines in make_symbols.sh like: 
cp line/wall=blocks ../../symbols/line/wall:blocks 

In Windows environment, ":" sign is illegal for file name. This leads to incorrect names like: "wall'blocks" which are not handled correctly lately in the zip files.

Is topodroid development restricted to Linux only? Is ":" sign mandatory for naming convention?

marcocorvi, 2017-04-17:
that's therion name for the symbol.

therion names were used whenever possible.
there is not a 1-1 map between topodroid symbols and therion symbols: both
programs are open to user-defined symbols

not sure, but i suspect it is not strictly necessary that filenames agree
with therion names.

marco

DSBork, 2017-04-17:
Great!
I have found that files with ":" sign in names are required for symbols.zip only. 
symbols.zip is already pushed to GIT, so I can avoid recreating it and use as is. 
Other symbol files were created successfully.
Now it works correctly.
Thank you.