marcocorvi/topodroid #29: Poll: should "line joining option stickyness" be increased?

rsevero, 2018-04-23:
I would like to know if current Topodroid users are satisfied with current "line joining option stickyness" level.

The line joining option I'm talking about is the one we have when we are drawing in Topodroid. The one that has the following options:
`none/start point/end point/both points/continue`

Currently if you change your drawing tool from whatever line type you were using when you set the "line joining option" to something else and them come back, the "line joining option" is reset no `none`.

I believe that the "line joining option" should stick so if I change it to something different than `none` it should be like that when I restart to draw lines.

What do you guys and girls think?

vldgeorgiev, 2018-04-23:
Hi

I agree completely. This is something that I have been thinking for a
while, but I haven't shared with Marco.

Vladimir

EQuirogaR, 2018-04-24:
I agree too. Topodroid should remember the selected option even if we change the line type until we change it again.

marcocorvi, 2018-06-07:
ok. will be in v. 4..0.1f 
(sorry for the late reaction to the issue)

closed 2018-06-07 by marcocorvi