marcocorvi/topodroid #35: Therion command "calibrate" - parser issue

randomtrue, 2018-12-21:
In my Therion import data from old measurments is command 

"calibrate compass -27.9" 

folowed by data line i.e

" 56 56.1 2.568 50.58 -7.38 "

Data format and declination is defined as

"declination 0.0 deg     
data normal from to length compass clino "

After recalibrated data I use the command to set calibration to initial state

"calibrate compass 0"

After import to the Topodroid, bearing value is Infinite or wrong value.

In Therion man is for 'calibrate' command only 'zero error' value obligator.
calibrate <quantity list> <zero error> [<scale>]

I think it is in the file ParserTherion.java
 
Issue location - ParserTherion.java rows 346-352 
Code:
                float zero = 0.0f;
                float scale = 1.0f;
                try {
                  scale = Float.parseFloat( vals[vals_len-1] );
                  zero  = Float.parseFloat( vals[vals_len-2] );
                } catch ( NumberFormatException e ) {
                  TDLog.Error( "therion parser error: scale/zero " + line );
                  zero  = scale;
                }

When the 'scale parameter' not used in command (the last value in line) exeption is triggered because the 'zero parameter' pointed to NAN value. Exception catch block set the 'zero parameter' to same value as 'scale parameter'. It is partly correct because 'scale parameter' still contains his value (wrong value).
In math at line 619-623 is descendant of 'scale parameter' used for correction of the bearing main value (divide it) from 'data line', but state.mScaleBer is set to wrong value (same value as state.mZeroBer).
If the code meet the command "calibrate compass 0", both parameter is set to zero, and infinity value generated.

Propose solutions:
When command is in shorter variant (must be correct), exeption is triggered on 'zero parameter' value parsing. In catch block is 'zero parameter' still zero, but 'scale parameter' contains value for it. When you add "if" command to catching this state, is posible to make clean the both parameters -> 'zero parameter' set to value carried on 'scale parameter' and at the end the 'scale parameter' set to 1.
Last test after all, is necessery to test the 'scale parameter' is not zero otherwise set it to 1.

marcocorvi, 2018-12-21:
replaced with
zero =0; scale = 1;
int k=1;
while ( k < vals_len - 1 ) {
   try { ++k; scale = ...; break; } catch ( IOException e ) { }
}
while ( k < vals_len - 1 ) {
  try { ++k; zero = ...; break; } catch ( IOException e ) { }
}

ie, scale gets the first float on the line, and zero gets the next float on the line (if any).

marcocorvi, 2018-12-21:
it will be in 4.1.1z and above

marcocorvi, 2018-12-22:
oops.
i meant: first try read zero, then scale

randomtrue, 2018-12-22:
Hi. Next problem is that bearing after correction goes under zero. All
negative bearings need add 360degree (or 400grad or full angle for other
units).
I looked to the Therion and output data (polygon projection) looks
different. I feel that correction of bearing has different direction.
Correction is pointing to the Nord (or zero value) therefore negative
correction value schuold be substracted from reading value (- - = +). If I
good remember the code of 'math', only swap 'zero parameter' and "scalet"
data value.

Best regards
Ladislav

randomtrue, 2018-12-22:
> therion book says on p.17
>
> measured = ( read - zero ) * scale
>
> i was in doubt whether "measured value" or "read value" is the "true
> value".
> my guess was that "measured" is what the instrument shows and "read" is the
> value that should be read.
>
> maybe it's the other way around

I mean this code (row 619?+)

  ber = state.mZeroBer + (ber*state.mUnitBer + state.mDeclination) /
state.mScaleBer;
                    } else {
                      ber = state.mZeroBer + (ber*state.mUnitBer) /
state.mScaleBer;
        }

marcocorvi, 2018-12-22:
yes, there are several issues:
(1) parsing zero/scale
(2) getting the "correct' formula to transform the number in the therion
file to the number to put in topodroid database
(3) enforce bounds on the values

(1) was buggy, and it is hopefully fixed with the double try-catch

(2) the semantics of topodroid is that numbers in the db are those measured
by an ideal distox, and the units are m for length, and degrees for angles,
the ideal distox measures the azimuth angles to the magnetic north

therefore manually entered numbers, like those that come from imported
files, can have correction transfirmation. it's just a matter to get
straight what this should be.
therion has zero, scale, and units.
it also handles declination because it is a cave mapping  program.
topodroid is a cave surveying program and does not treat declination
because it is not expected that the surveyor knows its value inside the
cave.
so what is this transformation ?

i got that i should use

topodroid_value = ( therion_value - therion_zero ) * therion_scale

now where do i put the therion_unit ?

and what about declination ?
if i got it correct, in therion data in the same survey file can have
different declination value,
how should this be handled ?

probably this issue is due to the fact that topodroid does not handle
declination, because it addesses only single survey trip, and it is not
expected that the declination changes *appreciably* in the time of a survey
trip

while on this issue:
should length calibration apply also to LRUD numbers ?

issue (3) is trivial
and you are right, angles should wrap around

thx for pointing out these issues about the parser of th files

marco

CaverBruce, 2019-03-29:
> therion book says on p.17 measured = ( read - zero ) * scale i was in doubt whether "measured value" or "read value" is the "true value". my guess was that "measured" is what the instrument shows and "read" is the value that should be read. maybe it's the other way around

A more complete Therion Book quotation is...
"calibrate <quantity list> <zero error> [<scale>] ▷ set the instrument calibration. The measured value is calculated using the following formula: measured value = (read value − zero error) × scale."

The terminology poorly chosen, but based on the way it is written and the arrangement of the formula, I would interpret it as 'measured value' is the true value that a perfectly calibrated instrument would return.  The 'read value' would be the value that is actually read from the instrument, and in the days of paper booking, would be written in the survey notes.

marcocorvi, 2019-10-24:
i believe this issue is closed - if there are follow ups do not hesitate to reopen it

closed 2019-10-24 by marcocorvi