marcocorvi/topodroid #36: XVI can not be read by xtherion

mdornseif, 2019-06-12:
**Describe the bug**

XVI Export can not be read by xtherion. Propbably because of missing `set XVIgrid` line.

**To Reproduce**
Steps to reproduce the behavior:
1. Settings -> Export -> Therion -> XVI on. See http://filez.foxel.org/f4bf9be412af for full config.
2. Export a Survey as ZIP.
3. unzip
4. try to open resulting `…0p.th2` file in xtherion (tried latest git revision 3f0396ee)
5. see error message regarding parsing: http://filez.foxel.org/1fa8308e19a6
6. If you remove the reference to `…0p.xvi` from the `…0p.th2` file can be opened. If you then try to insert the `…0p.xvi` you also get an parsing related error message. I was not able to determine the exact parsing error.
7. If you add something like `set XVIgrid {-1192.3228346456692 -2429.6456692913384 39.370078740157474 0.0 0.0 39.370078740157474 50 100}` at the end of the file XVI file all works as expected. 

**Expected behavior**
Survey including splay shots is displayed, like http://filez.foxel.org/1fa8308e19a6

**Device (please complete the following information):**
 - TopoDroid v. 4.1.3u

**Additional context**
The files generated by importing PocketTopo files contain the `set XVIgrid` line. Tested also with an older therion version but can't reproduce the exact version number.

Suggestion: Generate an `set XVIgrid` line. The format seems to be:

* minx
* miny
* grid size(?)
* 0.0
* 0.0
* grid size(?)
* (x points / grid size) + 1 ` int(($l_maxx - $l_minx) / $gsize) + 1`
* (y points / grid size) + 1

mdornseif, 2019-06-17:
Some XVI files are generated with a`set XVIgrid` line between `XVIstations` and `XVIshots`. So far, I can not discern a pattern when that happens.

marcocorvi, 2019-06-18:
I cannot get the linked files, but i think i got the problem.
When the xvi is exported it "must" include the grid.
This is currently controlled by a flag setting: when this is enabled the grid is included
in the xvi file and there should be no error.

The error is that the flag should not be considered and the gris always included.

In the next version this wil be hopefully fixed.

thanks for poynting out the error.