marcocorvi/topodroid #38: XVI Exports with commas instead of dots

mdornseif, 2019-07-19:
**Describe the bug**
Export of black XVI lines seems to have `,` as a decimal separator instead of `.`. This happens only to black lines. This is not readable by xtherion.

**To Reproduce**
Steps to reproduce the behavior:
1. Export a Sketch as XVI
2. Check toe XVI if it contains any `,`

**Expected behavior**
XVI export should be readable by xtherion and not contain strange decimal separators. 

**Screenshots**

```
  […]
  { red 315.37 93.85 319.15 92.21 }
  { red 319.15 92.21 322.56 91.60 }
  { red 322.56 91.60 325.87 91.48 }
  { black 250,50 94,65 250,50 118,28 }
  { black 250,50 106,47 262,32 112,37 }
  { black 250,50 106,47 262,32 94,65 }
  […]
```
**Device (please complete the following information):**
 - Galaxy Note 8
 - Android version 9
 - TopoDroid v 4.1.3u

mdornseif, 2019-08-06:
In new Exports I see even more commas:

```
-##XTHERION## xth_me_area_adjust 792.6 -433.1 1399.9 -164.2
+##XTHERION## xth_me_area_adjust 792,6 -433,1 1399,9 -164,2
 ##XTHERION## xth_me_area_zoom_to 25
-##XTHERION## xth_me_image_insert {196.85 1 1.0} {-236.22 0} windloch2019_g41-1p.xvi 0 {}
+##XTHERION## xth_me_image_insert {196,85 1 1.0} {-236,22 0} windloch2019_g41-1p.xvi 0 {}
```

Strange thing: I did not update Topodroid bewtween these Exports.

marcocorvi, 2019-10-24:
fixed in 5.0.0 - thanks

closed 2019-10-24 by marcocorvi