marcocorvi/topodroid #54: Importing a compass file (.dat) file does not work

Greenman126, 2023-01-24:
**Describe the bug**
When importing a simple .dat file via the import dialogue, nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Main Page
2. Click on Import Button
3. In the drop down menu, select Compass (if using dat file) or therion (if using th file)
4. No file is created

**Expected behavior**
I expect a file to be created or an error be thrown. Neither happens

**Device (please complete the following information):**
 - Samsung Active Tab2 and Samsung Galaxy s20FE
 - Android 9
 - TopoDroid 6.1.92-33

**Additional context**
No files with the same or a similar name to the dat file are used. DAT file attached (remove .txt)

[importtest.DAT.txt](https://github.com/marcocorvi/topodroid/files/10493749/importtest.DAT.txt)

marcocorvi, 2023-01-24:
"import" means to import an *existing* file, not to *create* a file.

see attached (length units m)

Greenman126, 2023-01-24:
I missed a step in my explanation.

I click import, I select "Compass" (or therion, can't get that to work either), then I click on a .DAT file I have stored on the device. When I click OK, the Import dialogue goes away and nothing happens

marcocorvi, 2023-01-24:
when you click ok it is done.
the survey has been imported : see screenshot previous mail

Greenman126, 2023-01-24:
Screenshot seems to be missing

marcocorvi, 2023-01-24:
it is a screenshot of the imported dat file:
just three shots a1-a2, and so on
azimuth and clino always 5°
lengths 1.52 16.76 and 1.52 m

closed 2023-01-31 by marcocorvi