marcocorvi/topodroid #61: No flowstone line. Unsupported rimstone-dam line (therion export)

Greenman126, 2023-02-06:
Therion has a useful linetype for "flowstone". This is absent in topodroid. Only "wall -subtype flowstone" is available in topodroid.

Also, a working tool for rimstone-dam is not present. There is a tool for gour-rim in topodriod, but it is not recognized in Therion. I believe a "gour-rim" should be titled "rimstone-dam" for export in therion.

marcocorvi, 2023-02-07:
well, you are welcome to write them (consider making a pull request on the
github repo).

the name should not be changed, for backward compatibility, and therion
names must be the same as filenames (program requirement).

marcocorvi, 2023-02-17:
waiting for pull request containing the new symbols

closed 2023-02-17 by marcocorvi