marcocorvi/topodroid #62: Can't change point size of new symbols

RaymondMichael, 2023-02-13:
1. In the sketching screen
2. Click on the drawing tool
3. Click in the lower right to bring up the option screen to select which thing to draw
4. Click on point symbols
5. Choose a point symbol
6. Change the size of the symbol to anything except M
7. Press the Back button to return to the sketching screen
8. Draw the symbol

With TopoDroid 5.x, the symbol would appear with the selected size. With TopoDroid 6.1.95 and 6.1.96, it appears with the Medium size. You can still select the symbol and change its size, but it would be better if you could create the point symbols with the correct symbol from the start.

marcocorvi, 2023-02-13:
thanks,

as a matter of fact, with the current drawing gesture, the size and
orientation controls in the symbol selection dialog are obsolete and ahould
be dropped.

marcocorvi, 2023-02-17:
fixed

closed 2023-02-17 by marcocorvi