marcocorvi/topodroid #65: Plan skecth shown when clicking at "Profile" in the skecths list: only in Portuguese

rsevero, 2023-02-28:
**Describe the bug**
When I try to open a profile sketch directly from the sketches list, the plan sketch is openned if the app is set to portuguese instead of the requested profile one. In english and french it works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Change the app language to portuguese
2. Open a survey with a scketch
3. Try to open a profile sketch directly clicking at "Perfil" (that is "profile" in portuguese)
4. You will see  the plan sketch and not the profile one you asked.

**Expected behavior**
the profile sketch should be presented and not the plan..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device (please complete the following information):**
 - Samsung Galasy 21 SE and Note 8
 - TopoDroid version 6.1.97-33

marcocorvi, 2023-02-28:
when a sketch is opened the app loads both plan and profile.
then displays accordingly to the tapped item.

it is strange that the choice between plan and profile fails in one
language and not in others.

i tried with italian, and it works as expected.

i suspect it is something related to the language files.
you should compile the app and add logs to see why it chooses plan instead
of profile, or run it under debug.

rsevero, 2023-02-28:
------- Original Message -------
On Tuesday, February 28th, 2023 at 3:25 PM, marco corvi ***@***.***> wrote:

> when a sketch is opened the app loads both plan and profile.
> then displays accordingly to the tapped item.
>
> it is strange that the choice between plan and profile fails in one
>
> language and not in others.

I agree completely.

> i tried with italian, and it works as expected.

Yes. As I said, in english and french it's also working.

> i suspect it is something related to the language files.

Possibily.

> you should compile the app and add logs to see why it chooses plan instead
> of profile, or run it under debug.

There was only one time, several years ago when I manage to compile and run Topodroid in my PC. I have tried again several times after that without sucess.

Recently I tried again but, as I don't really know Java and Android Dev SDK enough I never manage to import Topodroid's project in a way that is works in Android Dev SDK.

If you can send me a step by step guide on how to import Topodroid's project in Android Dev SDK, I would love to try to further debug this issue.Message ID: ***@***.***>

>

marcocorvi, 2023-02-28:
there is howto/build

for studio it is enough to make an empty project and link/copy 
AndroidManifest.xml, src, res, and assets. 
(you can do without native libs or comment their use )

rsevero, 2023-03-01:
Fixed with commit 6cf914914

_fmt_extended_ and _extended_ strings were different