marcocorvi/topodroid #67: Enable the "Point-wise/Multiselection/Join" action for areas

rsevero, 2023-03-05:
I have friends that wanted to be able to join areas of the same type in one single area.

I would like to use the "Point-wise/Multiselection/Join" action with 2 areas of the same type: this would join the begin point of one area to the end point of the other. In the end I would have only one single area.

marcocorvi, 2023-03-05:
i think that merging two areas is not supported.

rsevero, 2023-03-06:
It currently isn't in Topodroid. Can't it be? At the end there would be only one area with both areas joined.

marcocorvi, 2023-03-06:
why?

rsevero, 2023-03-06:
I have a friend that regulary needs to join water areas as usually he can't draw each lake in one single area. He ends overlapping different areas so they appear as one single water area.

marcocorvi, 2023-03-06:
this is something one does when he drafts the cave map.

why should it be useful in topodroid, drawing sketches during the cave
surveying trip ?

marcocorvi, 2023-03-06:
> I am not entirely sure about the answer to this issues you mentioned but I
> would bet that they would be the same if we were talking about
> "Point-wise/Multiselection/Join" action with lines which is implemented in
> Why TopoDroid would be able to join lines but not areas? I believe

this could create areas with disconnected borders.

topodroid areas have connected borders, ie, one closed (continuous) 2D
curve.

rsevero, 2023-03-06:
------- Original Message -------
On Monday, March 6th, 2023 at 9:36 AM, marco corvi ***@***.***> wrote:

> this could create areas with disconnected borders.
>
> topodroid areas have connected borders, ie, one closed (continuous) 2D
>
> curve.

Yes. You are right.

Unless this feature was restricted to the case where both areas actually intercept each other.

Well, in reality it's a little harder that I initially thought as eventually you even have to deal with the straight segments X bézier curves issue.

It would be nice to have it but it seems harder than I excepted.Message ID: ***@***.***>

>

marcocorvi, 2023-03-06:
The issue is more subtle than one might think at first.

Nevertheless i can conceive the need to "stretch" an area (for example, in the process of sketching the plan view) when a new leg is added to the survey and the area continues in the new leg.
i think that this can be more properly addressed with a "area continuation", similar to the "line continuation" function.
the area has a closed border, therefore it does not have all the different types of continuation as the line.

area continuation is not free of problems because one can continue an area to overlap with itself. i am undecided as to whether this is good or bad.

closed 2023-03-17 by marcocorvi