marcocorvi/topodroid #69: Hide from not working as expected: unlink shots still being shown

rsevero, 2023-03-18:
When I hide shots from a certain station I expected that all shots/stations "behind" it (by behind I mean, that where linked to the survey only through the hidden shot) are removed from the sketch but this is failing on the provided survey: https://we.tl/t-3uaBBEfbHO 

**To Reproduce**
Steps to reproduce the behavior:
1. Import the provided "Pratinha-4L" survey
2. Open the second plan sketch
3. At shot 4L21/4L22 hide shots from 4L21
4. See how we have many more shots sown than only the ones that are actually directly connected to the drawn sketch
5.  
**Expected behavior**
I expected that all other shots, disconnected by 4L21 were hdden (like 4l19, 4L49, 4L6 and so on)

**Screenshots**
[image]

**Device (please complete the following information):**
 - Samsung Note8
 - Android version: 9
 - TopoDroid version: 6.1.97-33

marcocorvi, 2023-03-18:
this is a long standing problem.
see comment in tdnum line 415.

marcocorvi, 2023-03-20:
probably fixed 6.2.1 next

closed 2023-03-20 by marcocorvi