marcocorvi/topodroid #72: Support to get data from new devices.

smgl9, 2023-05-02:
It is becoming more difficult to get a DistoX lately.

Hence support for new devices could be explored.
One option I've been thinking about is to get data directly from mobile. A device like DOOGEE S97 PRO has a telemeter incorporated and the other sensors needed. 

I guess that compass and accelerometer data is easy to get as they are integrated in almost every phone but don't know about the data from the telemeter.

marcocorvi, 2023-05-04:
> I will dare to share here the opinion of a friend, who had and tested one:
> Finally I've managed to make a test against DistoX. The results are not
> looking good. Actually this was to be expected.
> Try BRIC, it is already supported.

also:
the distoxble is in production (and one can still find leica x310).

guillaumechardin, 2023-05-04:
Hello Marco,
can you provide more details on DistoX-BLE, after a new search i found
something on your website, but the link is not working :
Main page : http://ww.marcocorvi.altervista.org/caving/notes/index.htm
Non working link : http://archive.org/details/distox-ble-assembly

thanks

marcocorvi, 2023-05-04:
swap x and -

fixed. tkx

guillaumechardin, 2023-05-06:
Hi Marco,
Can you tell me where the Distox2 BLE upgrade kit can be ordered ?

marcocorvi, 2023-05-07:
https://cavedevice.myshopify.com/products/cave-survey-tool

marco

smgl9, 2023-05-08:
I had the chance to test a mobile with telemeter. Disappointing results. 

As there isn't alternative devices for now, I guess I can close the issue.

closed 2023-05-08 by smgl9