marcocorvi/topodroid #73: Alternate palettes not appearing in Drawing Tools Selector

Greenman126, 2023-05-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' in the Top right of the starting page
2. Click on 'PALETTE'
3. No checkboxes or alternate tool symbols visible

**Screenshots**
[image]

**Device (please complete the following information):**
 - Samsung Active Tab2
 - Android 9.0
 - TopoDroid version 6.2.9-33

**Additional context**
This has happened in other versions and I'm not sure if it's a version issue or it just sometimes doesn't work for an odd reason.

rsevero, 2023-05-12:
Do you have the "Custom settings >> Palettes" option enabled?

Greenman126, 2023-05-12:
Yes, that is exactly the problem! I made the mistake of only turning on "Expert" mode when I should have turn on "Tester"

I recommend we move the alternate palettes option out of "Tester" and into "Normal". Seems like too important of a feature to hide.

closed 2023-05-12 by Greenman126