marcocorvi/topodroid #74: Retracing issue when drawing lines

Greenman126, 2023-05-12:
When drawing lines with retracing enabled, you cannot connect the end of a new line to the start of an old line. Attempting to do this, without retracing the entire wall, results in the older line being overwritten/destroyed.

Unless I am mistaken, this makes it impossible to connect lines to the origin point of old lines. This forces the user to retrace the entire old line every time they draw a new segment of the wall.

Is there a way around this? Is there a way to re-enable the old Line Connection functions?

marcocorvi, 2023-05-14:
retracing has been taken up by Mark Dickey - see issue #71
i am still waiting feedback from him.

marcocorvi, 2023-05-14:
> When drawing lines with retracing enabled, you cannot connect the end of a
> new line to the start of an old line. Attempting to do this, without
> retracing the entire wall, results in the older line [...]
retracing has been taken up by Mark Dickey,
see issue #71 on github.

> Is there a way to re-enable the old Line Connection functions?
no.
the code is no longer in the sources.

marco

marcocorvi, 2023-05-17:
the reported behavior could not be reproduced.

closed 2023-05-17 by marcocorvi

Greenman126, 2023-05-17:
Marco, I just tried this again (on a separate device too) and found that it only happens with bezier curves line type.

marcocorvi, 2023-05-17:
indeed that was what mark dickey had pledged to work on.

marcocorvi, 2023-05-17:
maybe i add a note in the manual that retracing does not work with bezier
and disable it for bezier lines

Greenman126, 2023-05-17:
Great. Thank you for being so responsive and for all your hard work Marco!

Greenman126, 2023-05-17:
Another suggestion: Give retracing its own toggle button.

I teach a lot of people about topodroid and I have difficulty explaining
how to enable and disable it. The long-press on the drawing tool button is
not very intuitive.

marcocorvi, 2023-05-17:
i considered that, but
(1) it is a frequent action thus it needs a button
(2) the number of buttons must be kept to a minimum.