marcocorvi/topodroid #75: Delete survey / shot should not be an 'advanced' feature.

RadostW, 2023-06-09:
**Problem**
I came back to using topodroid after quite some time and spent a day trying to figure out how to delete an accidental leg.

It seems to me that any newcomer tries to do a 'test' survey first and wants to be able to clean up errors etc.

**Cause**
By default only 'basic' features are enabled and user needs to enable 'advanced' features to be able to delete surveys and shots.
Additionally in all docs I've found the only guidance is 'just long press and delete will be there'

**Proposed solution**
Move delete actions from 'advanced' level to 'basic' level.

marcocorvi, 2023-06-09:
deleting shots is a "dangerous" operation because it can break the midline.

it can be done at "expert" level.
the default level is "normal".

i am not sure whether deleting shots should be available at a lower level
of activity.

marcocorvi, 2023-06-09:
sorry: "advanced" level, not "expert" level

RadostW, 2023-06-09:
In my case when DistoXBLE connects again with Bluetooth I sometimes get a bunch of rubbish shots that I don't want in my survey before I can take the actual shots measuring something.

If You're sure that You want to keep the current settings then can we have the information that delete is 'advanced' more easily available in the docs?

marcocorvi, 2023-06-09:
the xble should not transfer data that have been already sent.

why do you get these rubbish shots?

RadostW, 2023-06-09:
> the xble should not transfer data that have been already sent.

It has some old shots in memory and dumps them with the first shot. (From messing around on the surface, earlier survey where people did not use topodroid but just did paper etc).

It's also not that strange to make a poor measurement underground (that you want to remove)

marcocorvi, 2023-06-09:
> the xble should not transfer data that have been already sent.
>
> It has some old shots in memory and dumps them with the first shot. (From
> messing around on the surface, earlier survey where people did not use
> topodroid but just did paper etc).
you ought to have cleared the xble memory before you start the new survey.

> It's also not that strange to make a poor measurement underground (that
> you want to remove)
a poor shot can be renamed to be a splay, (or empty). it should be
commented specifying why.

wookey, 2023-07-10:
Marco. I don't believe that you should just close this issue and tell the users they are doing it wrong. Getting a load of shots from your disto when you open the software is absolutely normal. Users shouldn't have to know in advance that they won't be able to delete shots so will have to either start with a disposable test survey or find the magic button-runes to clear out the distoX shot store before connecting to the distoX.
Does anyone really use the software in 'basic' mode? I can't imagine never having to delete a shot in real surveying, especially when starting out with new software. Mistakes are guaranteed to be made on your first few surveys, and being able to just delete a shot is a really basic requirement. It seems to me that just about everybody must use this software in advanced (or expert) mode, because basic is nearly useless as soon as you actually try to do anything. Possibly once one has learned these various wrinkles one could go back to basic mode and use it successfully, but I know I've never tried. Do you really have users sticking with basic mode?
So far as I know, all users had to enable advanced mode almost immediately because they couldn't actually use it  without doing so (e.g. because of the need to delete duff legs).
I really do think that either basic mode should be abandoned, or shot deletion enabled in it, or at the very least making it easier to discover that one needs to turn on advanced mode in order to delete a leg.
Topodroid is famous for being hard to get to grips with. This is one simple thing that could be changed to make it a bit easier to get started with.