marcocorvi/topodroid #77: Option to automatically assign splay extend

Sjotroll, 2023-08-18:
**Is your feature request related to a problem? Please describe.**
Sometimes, when I take splays from the last station and looking forward into the cave which curves left or right, the splays are not EXTENDED but PROJECTED to the plane of the last leg (as described in the manual). In my experience I haven't encountered a situation where I would prefer the splays to be projected instead of extended (just like legs).

**Describe the solution you'd like**
Option to automatically set the extend of a splay to left or right based on the angle (+/- 90) relative to the previous leg. Or something like that, to disable projection of splays.
I understand that by extending and not projecting, all of the splays to be used for the plan view will be confusing in the section view, but this can be easily overcome by either: a) first taking splays for the section, sketching it, and then doing it for the plan, b) coloring the splays for section and plan in different colors, c) disabling plan splays from showing in the section view....

**Describe alternatives you've considered**
I've considered using "fixed extend reference", but haven't tried yet if it solves the problem.

**Additional context**
One side-issue of the unassigned extend of splays is that after taking the next leg, some of the projected splays from the last station are suddenly displayed as vertical, not sure what happens there. See the red splays in the figures (I manually colored the splays red to indicate the ones to be used for the extended section).
Note: the station policy I'm using is "backwards leg, splays".
[image]
[image]

marcocorvi, 2023-08-20:
> *Is your feature request related to a problem? Please describe.*
> Sometimes, when I take splays from the last station and looking forward
> into the cave which curves left or right, the splays are not [...]

the extented profile is the drawing on the cave on a "curtain" that follows
the midline.
therefore splays are "projected" on the pieces of this curtain.

> *Describe the solution you'd like*
> Option to automatically set the extend of a splay to left or right based
> on the angle (+/- 90) relative to the previous leg. Or something like that,
> to disable [...]

there are "splay classes": horizontal, vertical, transversal, normal.
but it is cumversome to use.

splay can also be set plan-only or profile-only.

> *Describe alternatives you've considered*
> I've considered using "fixed extend reference", but haven't tried yet if
> it solves the problem.

probably this does not help, because it concerns the legs.

> *Additional context*
> One side-issue of the unassigned extend of splays is that after taking the
> next leg, some of the projected splays from the last station are suddenly
> displayed as vertical, not [...]

the reason is that the new leg add another piece to the profile curtain.

it is preferable to use station policies with leg *after* splays

marcocorvi, 2023-08-28:
the implementation of automatic splay extend has been revised.
splay can be assigned an extend value.