marcocorvi/topodroid #78: Unable to compile

agseaton, 2023-12-20:
I'm having trouble building topodroid, would you be able to provide some guidance on how to do this?

I tried following the build instructions in `howto/build`, but neither of the two methods described worked for me.

Ant
---

I had to install an old version of the android SDK tools (25.2.5) and set the java.target and java.source values in ~/Android/Sdk/tools/ant/build.xml to 1.8. Otherwise I just got errors about an unsupported version.

Nevertheless, running `make` or `ant debug` gives me a bunch of errors:

```
Buildfile: /home/agseaton/Documents/topodroid/build.xml

-set-mode-check:

-set-debug-files:

-check-env:
 [checkenv] Android SDK Tools Revision 25.2.5
 [checkenv] Installed at /home/agseaton/Android/Sdk

-setup:
     [echo] Project Name: TopoDroidX
  [gettype] Project Type: Application

-set-debug-mode:

-debug-obfuscation-check:

-pre-build:

-build-setup:
[getbuildtools] Using latest Build Tools: 34.0.0
     [echo] Resolving Build Target for TopoDroidX...
[gettarget] Project Target:   Android 13
[gettarget] API level:        33
     [echo] ----------
     [echo] Creating output directories if needed...
    [mkdir] Created dir: /home/agseaton/Documents/topodroid/bin/rsObj
    [mkdir] Created dir: /home/agseaton/Documents/topodroid/bin/rsLibs
     [echo] ----------
     [echo] Resolving Dependencies for TopoDroidX...
[dependency] Library dependencies:
[dependency] No Libraries
[dependency] 
[dependency] ------------------
     [echo] ----------
     [echo] Building Libraries with 'debug'...
   [subant] No sub-builds to iterate on

-code-gen:
[mergemanifest] No changes in the AndroidManifest files.
     [echo] Handling aidl files...
     [aidl] No AIDL files to compile.
     [echo] ----------
     [echo] Handling RenderScript files...
     [echo] ----------
     [echo] Handling Resources...
     [aapt] No changed resources. R.java and Manifest.java untouched.
     [echo] ----------
     [echo] Handling BuildConfig class...
[buildconfig] Generating BuildConfig class.

-pre-compile:

-compile:
    [javac] Compiling 671 source files to /home/agseaton/Documents/topodroid/bin/classes
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/TopoDroidApp.java:51: error: cannot find symbol
    [javac] import com.topodroid.common.ExportInfo;
    [javac]                            ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: package com.topodroid.common
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SurveyWindow.java:22: error: cannot find symbol
    [javac] import com.topodroid.common.ExportInfo;
    [javac]                            ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: package com.topodroid.common
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/TopoDroidApp.java:3119: error: cannot find symbol
    [javac]   boolean doExportDataAsync( Context context, ExportInfo export_info, boolean toast )
    [javac]                                               ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: class TopoDroidApp
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SurveyWindow.java:610: error: cannot find symbol
    [javac]   private static ExportInfo mExportInfo; // index of the export-type 
    [javac]                  ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: class SurveyWindow
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SaveDataFileTask.java:21: error: cannot find symbol
    [javac] import com.topodroid.common.ExportInfo;
    [javac]                            ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: package com.topodroid.common
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SaveDataFileTask.java:46: error: cannot find symbol
    [javac]   private ExportInfo mExportInfo;
    [javac]           ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: class SaveDataFileTask
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SaveDataFileTask.java:62: error: cannot find symbol
    [javac]   SaveDataFileTask( Uri uri, String format, long sid, SurveyInfo info, DataHelper data, String survey, Device device, ExportInfo export_info, boolean toast )
    [javac]                                                                                                                       ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: class SaveDataFileTask
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SurveyWindow.java:591: error: cannot find symbol
    [javac]         mExportInfo = new ExportInfo( index, prefix, filename );
    [javac]                           ^
    [javac]   symbol:   class ExportInfo
    [javac]   location: class SurveyWindow
    [javac] /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/SaveDataFileTask.java:113: error: illegal parenthesized expression
    [javac]         switch ( mExportInfo.index ) {
    [javac]                ^
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: /home/agseaton/Documents/topodroid/src/com/topodroid/TDX/Archiver.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 9 errors

BUILD FAILED
/home/agseaton/Android/Sdk/tools/ant/build.xml:716: The following error occurred while executing this line:
/home/agseaton/Android/Sdk/tools/ant/build.xml:730: Compile failed; see the compiler error output for details.

Total time: 4 seconds
```

Android studio
-----

The setup form for an empty project has changed, so I wasn't able to enter the parameters you specified. But I was able to configure most of these options in the build.gradle file.

I then attempted to build and got the following error:
```
Setting the namespace via the package attribute in the source AndroidManifest.xml is no longer supported.
```

Removing the package attribute fixed that, but now I get:
```
Android resource linking failed
warn: removing resource com.example.TDX:array/lineContinue without required default value.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:37: error: resource color/purple_200 (aka com.example.TDX:color/purple_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:38: error: resource color/purple_700 (aka com.example.TDX:color/purple_700) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:39: error: resource color/black (aka com.example.TDX:color/black) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:41: error: resource color/teal_200 (aka com.example.TDX:color/teal_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:42: error: resource color/teal_200 (aka com.example.TDX:color/teal_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:43: error: resource color/black (aka com.example.TDX:color/black) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:37: error: resource color/purple_200 (aka com.example.TDX:color/purple_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:38: error: resource color/purple_700 (aka com.example.TDX:color/purple_700) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:39: error: resource color/black (aka com.example.TDX:color/black) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:41: error: resource color/teal_200 (aka com.example.TDX:color/teal_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:42: error: resource color/teal_200 (aka com.example.TDX:color/teal_200) not found.
com.example.TDX.app-mergeDebugResources-26:/values-night-v8/values-night-v8.xml:43: error: resource color/black (aka com.example.TDX:color/black) not found.
error: failed linking references.
```

marcocorvi, 2023-12-20:
Ant
exportinfo is not yet under git.

you can try either (1) an old commit or (2) write the class (which is a struct) by looking at the points in the code where it is referred to (see javac messages), or (3) wait for a next commit.

studio
i have not tried it in months - studio builds takes too much more time than ant.
from the messages it seems you left out colors.

agseaton, 2023-12-20:
Okay, thanks for the suggestions. I will give these a try and get back to you when I have the opportunity.

Incidentally, have you considered setting up a continuous integration tool? I think it would be very beneficial to ensure that commits that are submitted to the repo do not cause the build process to fail.

marcocorvi, 2023-12-20:
no CI because compiling needs the android development environment and traslation commits often need fixing.

the issue is closed because you were not definite about trying to compile.

i  always check that the code compiles before a commit.

closed 2023-12-20 by marcocorvi

agseaton, 2023-12-20:
> no CI because compiling needs the android development environment and traslation commits often need fixing.

I'm not sure I understand. I don't see why the android development environment couldn't be included in a CI setup.

> i always check that the code compiles before a commit.

And yet the code doesn't compile with the latest commit...

> the issue is closed because you were not definite about trying to compile.

Okay...   ...as I said, I will give your suggestions a try and get back to you.