marcocorvi/topodroid #79: Manually setting active statio

rsevero, 2023-12-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
If applicable, a clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device (please complete the following information):**
 - Make and model
 - Android version
 - TopoDroid version

**Additional context**
Add any other context about the problem here.

marcocorvi, 2023-12-31:
this is the expected behaviour.
active station is applies only to legs, 
splays assignment is by the  policy

rsevero, 2023-12-31:
For policy "Splays, forward legs" splay assignement is by "active station"

For policy "backward legs, splays" splay assignement is by "policy"

Besides this difference in behaviour that I believe shouldn't have because of different policies, I really don't understand why splay assignment wouldn't respect "active station". I have always used it. It is really usefull. Yesterday I found out that "backward legs, splays" behaves, in this aspect, differently to "Splays, forward legs".

marcocorvi, 2023-12-31:
then it must be an error in the policy code.
i wiil check it.

marcocorvi, 2023-12-31:
the same behaviour was in the policy "first forward-leg then splays"

somehow there was a comment in the code that said it was to be like that,
and bypassed the check of the active station.

i do not remember why i put it,
and i am not sure that adding the check there might be some drawback.

rsevero, 2024-01-05:
With this file with TopodroidX 6.2.35 and a BRIC4 I have 2 issues:

1. setting active station still ignored, i.e., new splays created from last created station, ignmoring "active station" setting
2. turning off "legs-only refresh" inefective: to see new splays I have to go to shots list and them back to drawing window.

[teste2.zip](https://github.com/marcocorvi/topodroid/files/13840388/teste2.zip)

marcocorvi, 2024-01-05:
Only bric4 or also distox2 ?

rsevero, 2024-01-05:
> Only bric4 or also distox2 ?

Again the tests that failed were with a BRIC4. Forgot to test with disto.

marcocorvi, 2024-01-05:
the wrong behavior could not been reproduced 
v. 6.2.35

rsevero, 2024-01-05:
Yes. It seems to be working now. Strange. If I manage to get more solid info I wil get back to you. Sorry for the noise.

marcocorvi, 2024-01-07:
since it "seems" to be working i close the issue.