marcocorvi/topodroid #80: Stylus only drawing being ignored

rsevero, 2023-12-31:
**Describe the bug**
Stylus only drawing stopped working: any setting different from 0 makes drawing motions to be ignored, i.e., nothing is drawn.

**To Reproduce**
Steps to reproduce the behavior:
1. Set stylus only drawing to any value different from 0
2. Try to draw with a stylus that you previously has used to draw with "stylus only draw" set to 1

**Expected behavior**
You should be able to draw but nothing is drawn

**Device (please complete the following information):**

- Samsung Galaxy S21 FE
- Android version 13
- TopoDroid version 6.2.34-33

marcocorvi, 2023-12-31:
check the size of the stylus

rsevero, 2023-12-31:
Just checked again: from 1 to 5 stylus only setting I can't draw neither with the stylus or with finger. With 0 both work.

marcocorvi, 2023-12-31:
non-positive means disabled.
increase the size of the stylus

rsevero, 2023-12-31:
> non-positive means disabled. increase the size of the stylus

I know. I tried positive numbers. From 1 to 5 both the stylus and my finger are now ignored.

marcocorvi, 2024-01-07:
the size is in pixels.

rsevero, 2024-01-07:
> the size is in pixels.

I tried way bigger values up to 40 and still can't draw neither with my stylus nor with my finger when setting "Stylus only drawing" != 0

marcocorvi, 2024-01-07:
try a really large integer.

rsevero, 2024-01-07:
> try a really large integer.

Tried 400 and 4000. Still nothing.With zero both stylus and finger work.

marcocorvi, 2024-01-07:
this is wierd:

4000 is likely more than the display size.
the size (major axis) of the finger touch should be smaller than that.

my android display has a resolution about 441 ppi and the touch of a finger
tip is usually 40 pixels or more, but can be made almost comparable to a
fat-tip stylus.

i recently tested the stylus-only mode again, and had no problem.

rsevero, 2024-01-07:
I am at a loss also here. I used before in some tests. If you can think of anything I might test here to help understand, please let me know.

marcocorvi, 2024-01-07:
get 6.2.37-34
then do "adb logcat" and filter on "topodroid-X"

rsevero, 2024-01-07:
> get 6.2.37-34 then do "adb logcat" and filter on "topodroid-X"

Sorry for the silly question but, how to I "do adb logcat"?

marcocorvi, 2024-01-07:
install adb on your pc.
enable usb debug on android.
connect pc to android with usb.
execute adb logcat on pc,
and filter on topodroid-X.

rsevero, 2024-01-07:
Setting: 80

With the stylus:
```
01-07 08:48:48.815 13061 13061 V topodroid-X: STYLUS tool 0 size 0.0627451 major 6.6796875
01-07 08:48:48.839 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:48.855 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:48.889 13061 13061 V topodroid-X: STYLUS tool 0 size 0.068627454 major 7.932129
01-07 08:48:48.906 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:48.955 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:48.972 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:48.989 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.006 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.022 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.039 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.056 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.106 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.139 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.206 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.222 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:49.239 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:49.256 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07058824 major 8.349609
01-07 08:48:49.272 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.289 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.306 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07254902 major 8.76709
01-07 08:48:49.322 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.339 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.356 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.372 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.389 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.406 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.422 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.439 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.456 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 9.602051
01-07 08:48:49.472 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.489 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.505 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.522 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.539 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.556 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.572 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.589 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.606 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.622 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.640 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.656 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.672 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.689 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 9.18457
01-07 08:48:49.771 13061 13061 V topodroid-X: STYLUS tool 0 size 0.0627451 major 6.6796875
01-07 08:48:49.771 13061 13061 V topodroid-X: STYLUS tool 0 size 0.0627451 major 6.6796875
01-07 08:48:49.771 13061 13061 V topodroid-X:  Bezier 4 -> 1
```

With my finger:
```
01-07 08:48:54.115 13061 13061 V topodroid-X: STYLUS tool 0 size 0.0627451 major 6.6796875
01-07 08:48:54.156 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:54.172 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:54.189 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:54.206 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:54.222 13061 13061 V topodroid-X: STYLUS tool 0 size 0.064705886 major 7.097168
01-07 08:48:54.239 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07450981 major 7.932129
01-07 08:48:54.256 13061 13061 V topodroid-X: STYLUS tool 0 size 0.07647059 major 8.349609
01-07 08:48:54.272 13061 13061 V topodroid-X: STYLUS tool 0 size 0.078431375 major 8.349609
01-07 08:48:54.289 13061 13061 V topodroid-X: STYLUS tool 0 size 0.08039216 major 8.76709
01-07 08:48:54.306 13061 13061 V topodroid-X: STYLUS tool 0 size 0.08235294 major 8.76709
01-07 08:48:54.322 13061 13061 V topodroid-X: STYLUS tool 0 size 0.08235294 major 8.76709
01-07 08:48:54.339 13061 13061 V topodroid-X: STYLUS tool 0 size 0.08431373 major 9.18457
[... cut ...]
01-07 08:48:55.022 13061 13061 V topodroid-X: STYLUS tool 0 size 0.10392158 major 11.271973
01-07 08:48:55.039 13061 13061 V topodroid-X: STYLUS tool 0 size 0.10392158 major 11.271973
01-07 08:48:55.056 13061 13061 V topodroid-X: STYLUS tool 0 size 0.10392158 major 11.271973
01-07 08:48:55.089 13061 13061 V topodroid-X: STYLUS tool 0 size 0.10196079 major 11.271973
01-07 08:48:55.146 13061 13061 V topodroid-X: STYLUS tool 0 size 0.10196079 major 11.271973
01-07 08:48:55.147 13061 13061 V topodroid-X:  Bezier 39 -> 3
```

marcocorvi, 2024-01-07:
there should be other messages before these.

rsevero, 2024-01-07:
Log file: https://we.tl/t-07WFSHZfVp

marcocorvi, 2024-01-07:
since the setting is 80 and the size in 8-9 for stylus and 10-11 for finger
tip, lines should be traced with both

marcocorvi, 2024-01-07:
> Unfortunatelly they aren´t. Only if I set to zero.

do you see the white trace?

rsevero, 2024-01-07:
Just a tiny piece at the beggining of the line, right after I touched the screen but as I keep drawing the line, only that small part of white line is there.

When I remove the stylus from the screen, the tiny bit of white line dispappears and no line is actually drawn.

rsevero, 2024-01-07:
> try 6.2.38 filter on "STYLUS"

When I draw really slow, it works. Both with the stylus and with my finger. 

I never tried drawing slowly like this in previous versions.

rsevero, 2024-01-07:
Forgot to filter. Sorry. Just a moment

rsevero, 2024-01-07:
[topodroidx-stylus.log](https://github.com/marcocorvi/topodroid/files/13853488/topodroidx-stylus.log)

marcocorvi, 2024-01-07:
what is your line-point spacing ?

rsevero, 2024-01-07:
> what is your line-point spacing ?

It was 6 (don´t remember why I changed to 6). Changed back to 10 and everything seems fine.

It is strange that having the line spacing at 6 is a problem only with stylus size different from zero.

marcocorvi, 2024-01-07:
stylus-only is suspended when the stylus is traced much faster than the
line point spacing.

the difference between 6 and 10 is small.
right now "faster" is slightly more than "trice".

i do not remember why the suspending.
i will add a note to the manual.
and increase "faster " a bit.

rsevero, 2024-01-07:
> stylus-only is suspended when the stylus is traced much faster than the line point spacing. the difference between 6 and 10 is small. right now "faster" is slightly more than "trice". i do not remember why the suspending. i will add a note to the manual. and increase "faster " a bit.

On my other, slower, phone, even with spacing 10 (and stylus not zero), I have to draw slower than usual.

marcocorvi, 2024-01-07:
try 6.2.39
please.

rsevero, 2024-01-07:
> try 6.2.39 please.

It is working great on both phones. Thanks!!

marcocorvi, 2024-01-07:
the bound with the spacing is important
it is a workaround to prevent spurious artifacts.

closed 2024-01-07 by marcocorvi