marcocorvi/topodroid #82: "Line join" button only available after forcing app to stop when turning on "Settings > Sketching > Lines > Line join"

rsevero, 2023-12-31:
**Describe the bug**
"Line join" button only available after forcing app to stop when turning on "Settings > Sketching > Lines > Line join"

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on "Settings > Sketching > Lines > Line join"
2. Start to draw with "wall" line type

**Expected behavior**
The "line join" button should be available but I only got it after leaving TopoDroidX, forcing it to stop and them coming back.

**Device (please complete the following information):**
- Samsung Galaxy S21 FE and Motorola A32
- Android version 13
- TopoDroid version 6.2.34-33

marcocorvi, 2023-12-31:
handling of the setting was in the wrong group: GEEKLINE instead of LINE
hopefully fixed in the next version.

closed 2023-12-31 by marcocorvi

rsevero, 2023-12-31:
First of all, thanks for your prompt answer. It's great to use your software.

The link for the new version (6.2.35) is broken. Can uyou take a look when possible?

And happy new year!!

marcocorvi, 2023-12-31:
right!
topodroid is now built with 34.
thx.