marcocorvi/topodroid #85: When making a station active in drawing window, sometimes the station isn't drawn in green

rsevero, 2024-01-04:
**Describe the bug*
When making a station active in the drawing window, sometimes the station isn't drawn in green

**To Reproduce**
Steps to reproduce the behavior:
1. Make a station active in the drawing window
2. *Sometimes* the station isn't drawn in green.

**Expected behavior**
For the newly actived station be always drawn in green. 

**Additional context**
The choosen station is actually actived every time. It's just the "drawn in green" functionality that is erratic in the drawing window.

rsevero, 2024-01-05:
I am trying to identify what events makes this behaviour to start. I already know that closing TopoDroidX AND forcing it to stop in Android makes it stop.

marcocorvi, 2024-01-05:
android model and os version ?

rsevero, 2024-01-05:
> android model and os version ?

-     Samsung Galaxy S21 FE 
-     Android version 13
-     TopoDroid version 6.2.34-33

rsevero, 2024-01-05:
> current version is 6.2.35

Sorry. I am using 6.2.35. the problem remains on 6.2.35

rsevero, 2024-01-05:
When testing for "Legs-only refresh" issue, I activated a station on the drawing. It didn´t get green. I made a measurement anyway: the splay got attached to the right (activated but not green) station. When the new splay was drawn, the station became green.

marcocorvi, 2024-01-20:
it could not be reproduced (v. 6.2.42 and later)
closed by lack of further feedbacks

closed 2024-01-20 by marcocorvi