marcocorvi/topodroid #86: Projected profile direction window very slow

rsevero, 2024-01-24:
**Describe the bug**
When creating a projected profile in a Motorola g42 the window is very slow to the point it is not usable. On a Galaxy 23fe it works fine. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached survey
2. Go to new scrap
3. Choose projected profile. 
4. Try to choose a direction for the projected profile 
5. See how the image movements are not responsive to the point it is unusable

**Expected behavior**
For the project profile direction setting window image to move in a responsive way in both cel phones. 

**Device (please complete the following information):**
 - Motorola g42 (bad) Samsung Galaxy 23fe (good) 
 - Android 13
 - TopoDroid version 6.2.43

[teste.zip](https://github.com/marcocorvi/topodroid/files/14031608/teste.zip)

marcocorvi, 2024-01-24:
hardware problem ?

try using the -/+ buttons that change azimuth one degree at a time

in 6.2.45 the -/+ buttons respond to long click to change azimuth by 10 degrees

marcocorvi, 2024-02-13:
closed because there was no follow up

closed 2024-02-13 by marcocorvi