marcocorvi/topodroid #91: Downloading measurements from BRIC4 with option "Read all" and having repeated measurement IDs fails without proper error message

rsevero, 2024-02-12:
**Describe the bug**
Downloading measurements from BRIC4 with option "Read all" and having repeated measurement IDs fails without proper error message

**To Reproduce**
Steps to reproduce the behavior:
1. Set "Custom sttings > Devices > BRIC mode" to "All"
2. Download some measurements from BRIC
3. Clear BRIC memoery
4. Make new measurements and try to download them from BRIC
5. See how Topodroid won't download, neither give some proper merror message. It will only hang tryng to download the measurements

**Expected behavior**
At least present  an informative error message like "Can't download measurements with repeated IDs". Eventually offer to change the "BRIC mode" option to "All except index"

**Device (please complete the following information):**
 - Samsung 21 FE 
 - Android 13
 - TopoDroid 6.2.51

marcocorvi, 2024-02-13:
there are two different options: "all with index" and "all except" index.

tried both.
neither could reproduce the error.

please provide more information

rsevero, 2024-02-13:
Yesterday, inside the cave I couldn't download new data from my BRIC4 and only managed to download it after changing the BRIC4 mode setting from All to All except index. That's why I filed this issue.

I can't reproduce it here also. Sorry for the noise.

closed 2024-02-13 by rsevero

marcocorvi, 2024-02-13:
this was a real issue.
the default data mode for the bric is to use the bric index as shot id. if the bric memory is cleared, the index restarts from 1. the new shots are id-duplicate of other shots that are already in the database. since the shot list is ordered by the id, they are not appended at the end. 
from v 6.2.52 this changes to override bric index if it conflict with the existing survey data.'