marcocorvi/topodroid #92: When importing zip files with wrong filename, offer to try to import ignoring filename

rsevero, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
It's frustrating when trying to import zip files in Topodroid and get a "invalid manifest file" only because the filename was changed from _Survey_name.zip_ to _Survey_name (1).zip_ or something similar

**Describe the solution you'd like**
Topodroid could present the expected name available in the manifest file and offer ot import despite the zip filename not being the same as the one in the manifest file.

marcocorvi, 2024-02-13:
requirement that survey name matches zip name was not enforced in the code
for a long time.

rsevero, 2024-02-13:
Humm, maybe the person that reported that to me did something else with the zip file. I will check. Sorry for the confusion.

closed 2024-02-13 by rsevero

Tapter, 2024-02-13:
I once noticed the same behavior back in topodroid 6.1, however I haven't seen it happening on 6.2 - probably it has been fixed since