marcocorvi/topodroid #94: Name an station as "2E..." makes Topodroid produce errors continously

rsevero, 2024-02-12:
**Describe the bug**
Name an station as "2E..." makes Topodroid produce errors continously

**To Reproduce**
Steps to reproduce the behavior:
1. Name a station as "2E..."
2. Try to download new shots
3. See Topodroid hang with tons of errors

**Expected behavior**
Topodroid should only acept digits and lettters in station names.

**Device (please complete the following information):**
 - Motorola g42
 - Android 13
 - TopoDroid 6.2.51

marcocorvi, 2024-02-13:
station names terminating with non-alphanumeric characters can hang the
search for the next name.

therefore  a check will be implemented that forces station names to end
with an alphanumeric character.

it can reject the name, since a bad name is probably due  to a user mistake

rsevero, 2024-02-13:
In my case it wasn't a mistake but if the name were rejected I would have named something else.

I believe you can only reject names with any chat not digit or letter.

closed 2024-02-13 by marcocorvi