marcocorvi/topodroid #95: Set splays color at shot list window

rsevero, 2024-04-01:
**Is your feature request related to a problem? Please describe.**
It would make the splay colour feature much more usefull to me if I could set splay color at the shot list window so, for example, I could color the last splays that I made just for a cross section.

**Describe the solution you'd like**
Both in the single splay edit window and at the multi splay selection, ther should be a way to set the splay (or splays) color.

**Describe alternatives you've considered**
I tried to color the splays at the plan or profile view but it gets hard to identify which splays I want to color. In the shots list window it's really easy (usually the X last ones).

closed 2024-04-01 by marcocorvi

rsevero, 2024-04-01:
How can I set splay color on multi shot selection? I can't fix it.

marcocorvi, 2024-04-02:
must be some setting.
a document listing all settings and their effects, and relation among them.