marcocorvi/topodroid #96: "Refresh" button separated by empty space from the other buttons at "draw section" window

rsevero, 2024-04-01:
**Describe the bug**
"Refresh" button separated by empty space from the other buttons at "draw section" window

**To Reproduce**
Steps to reproduce the behavior:
1. Set "test" resource level
2. Go to "draw section" window
3. See that there is an empty space between the "refresh" button and the other buttons.

**Expected behavior**
The "refresh" button should have no empty space separating it from the other buttons.

**Screenshots**
[image]

**Device (please complete the following information):**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.57-34

marcocorvi, 2024-04-01:
removed refresh button from the interface.
it is not very important for x-section

rsevero, 2024-04-01:
> removed refresh button from the interface. it is not very important for x-section

Really!?

I use it a lot as new plays never appear automatically on X-sections.

marcocorvi, 2024-04-02:
no need to reopen: you should have checked the sources.
there is a set of buttons specific for x-sections, now.

closed 2024-04-02 by marcocorvi