marcocorvi/topodroid #99: Update the symbols in the next release

rsevero, 2024-04-03:
**Is your feature request related to a problem? Please describe.**
I have made a few symbol updates recently (one today, a few this year) that are not yet reflected in Topodroid releases. Can you please update symbols on the next release?

marcocorvi, 2024-04-04:
done

closed 2024-04-04 by rsevero

rsevero, 2024-04-04:
You didn´t include the last change I made on low-ceiling line. Can you include it please?

marcocorvi, 2024-04-04:
strange.
i pulled from github, than i made the symbols zips. i increased the symbol
version to 42, and i build the apk.

in which set it your symbol?

rsevero, 2024-04-04:
> strange. i pulled from github, than i made the symbols zips. i increased the symbol version to 42, and i build the apk. in which set it your symbol?

42.

BTW, it would be nice if the symbol set version was also presented on the dialog box where we choose which symbol sets should be installed.

Right now it's only presented on the simplier install/update symbol set dialog box.

rsevero, 2024-04-04:
> i meant: which class of symbols?

speleo_extra

marcocorvi, 2024-04-04:
lowceiling is included in the zip of symbol extra on github, therefore it
should be in the apk.
if there is a syntax error in the file it is not loaded at runtime.
this error should be in the syslog.

rsevero, 2024-04-04:
> lowceiling is included in the zip of symbol extra on github, therefore it should be in the apk. if there is a syntax error in the file it is not loaded at runtime. this error should be in the syslog.

The file is there. But it's the old version. Not the updated one.

marcocorvi, 2024-04-04:
maybe i made the zip before pulling your update to low ceiling.

i will make v. 43

marcocorvi, 2024-04-04:
i don't undestand:

the dialog to install/update is where you choose which symbol should be
installed.

this is the dialog that shows the version.
what is the other dialog?

rsevero, 2024-04-04:
> i don't undestand: the dialog to install/update is where you choose which symbol should be installed. this is the dialog that shows the version. what is the other dialog?

The "Pallete" menu option in the main window shows 2 different dialog boxes, depending on the resource level being set to "Tester" or something lower.

If it is "Tester", if presetns the more complex dialog box that does not show the symbol set version.

If the resource level is anything less than "Tester", it presents the simplier version of the dialog box. This one shows the symbol set version.

marcocorvi, 2024-04-04:
i see.

yes, the text with version numbers can be  shown at any level of activity.

i change that.