marcocorvi/topodroid #100: "More shot edit" window has no "Ok" button

rsevero, 2024-04-04:
**Describe the bug**
"More shot edit" window has no "Ok" button

**To Reproduce**
Steps to reproduce the behavior:
1. Go to shot edit dialog box for a splay.
2. Click on "More" to open the "More shot edit" window
3. Enter LRUD data or change the splay color.
4. See that there is no "Ok" or "Save" button to close and save the changes.

**Expected behavior**
There should be a "Ok" or "Save" button on the "More shot edit" window so the user can save any changes.

**Device (please complete the following information):**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.60-34

marcocorvi, 2024-04-04:
lrud are at an intermediate point, for old-style cave models.
therefore, they should not be there for splay shots.

all buttons are actions after which the dialog is closed (in practice the
dialog may even close before the action is performed).
when the action is done the control returns to the data list window.

rsevero, 2024-04-04:
> lrud are at an intermediate point, for old-style cave models. therefore, they should not be there for splay shots. all buttons are actions after which the dialog is closed (in practice the dialog [...]

Not for the color setting.

marcocorvi, 2024-04-04:
it is just that the color is not imnediately shown in the list.
this is stated in the help page

rsevero, 2024-04-04:
> it is just that the color is not imnediately shown in the list. this is stated in the help page

When I set the splay color using the color option in the "More shot edit"  dialog, after setting the color in the color palete, I am returned to the "More shot edit" dialog box. The color is already set but I only have the "Cancel" button to close the window. 

As you said, the color is not immediatly shown. It gets really confusing for the user. I supposely set the color but I clicked in a "Cancel" button and the color in not shown. Is the color really set or not?

I see that it is but it is really confusing. Maybe if the whole dialog box was closed when I choose the color in the color palete it woudl get less confusing.

And, even better, if the new color is shown at this moment in the splay, then it would be perfect.

marcocorvi, 2024-04-04:
i agree with you:
the color immediately shown is the best.

i did not do it, because i was counting the number of taps to set color.
with the dialog it is 3: open the edit dialog, the more dialog, the color.
with multishot it is 3: go multishot, tap the light button, then the color.

i'm not sure that having the color in the more dialog is any better that
using the multishot

rsevero, 2024-04-04:
> i agree with you: the color immediately shown is the best. i did not do it, because i was counting the number of taps to set color. with the dialog it is 3: open the edit dialog, the more dialog, [...]

I agree that using the multishot color editing is better but not having it at the shot editor is confusing. I really believe it should be there also.

But if you prefer to remove it, I can leave without it. Newbies might have a harder time.

marcocorvi, 2024-04-04:
newbies don't go to the more dialog.

one big problem of topodroid is how to make the interface simpler.
adding more stuff inevitably makes the interface more complex.

the more dialog is a catch for all those wierd seldom used actions, that
would make the edit dialog unnecessary cluttered.

as a matter of fact "more" should be "more/less" and open/close the box
with  extra actions.
i never liked having two dialogs.
at first it was necessary because of screen estate, and scrolling in a
dialog is not nice, but now phones are becoming taller.

rsevero, 2024-04-04:
> newbies don't go to the more dialog. one big problem of topodroid is how to make the interface simpler. adding more stuff inevitably makes the interface more complex. the more dialog is a catch for [...]

I like this idea.

marcocorvi, 2024-04-16:
Shot edit-more dialog has been included in shot edit dialog

closed 2024-04-16 by marcocorvi