marcocorvi/topodroid #101: App not respecting "Language" setting

rsevero, 2024-04-05:
**Describe the bug**
App not respecting "Language" setting.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings and change language setting
2. See that the texts on the current page are translated to the new setting but other pages no.

**Expected behavior**
Expected that all pages texts appear in the new language choosen.

**Screenshots**
See **Additional context** below:
[image]

**Device (please complete the following information):**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.60

**Additional context**
After I leave the app and force stop it, all texts are back in the system default language (see the screenshot above).

rsevero, 2024-04-05:
Apparently the problem is just on the settings pages. The rest of the app follows the language setting.

marcocorvi, 2024-04-06:
can you describe better?

rsevero, 2024-04-06:
> can you describe better?

Have you seen the first message in this issue?

marcocorvi, 2024-04-06:
yes, and i cannot reproduce it.
maybe i do a different set of actions

rsevero, 2024-04-07:
Which is the default language of the Android devices you are testing? Mine is portuguese.

Just tested that the main settings page returns to the default Android language after you get out of the settings page to the main TopoDroidX page and back.

After changing the language setting in my Adnroid to english, the main settings page is always in english despite the settings I chose.

marcocorvi, 2024-04-07:
my default is english.

when i set language italian i get everything in italian
when i set default i get everything in english

rsevero, 2024-04-08:
Currently I have this issue in my 2 cel phones. In my "dev" install on Android Studio the language setting is completely ignored.

Can you give me some pointers on where should I start looking to try to solve this issue?

marcocorvi, 2024-04-08:
locale is set with a configuration change

marcocorvi, 2024-04-09:
i tested after setting italian as default language.
when i select english it takes a few seconds before the interface is
updated to english.
that time is probably necessary to load the translation in the runtime.

if i then set to default it returns to italian, again after some seconds.

rsevero, 2024-04-09:
> i tested after setting italian as default language. when i select english it takes a few seconds before the interface is updated to english. that time is probably necessary to load the translation in the runtime. if i then set to default it returns to italian, again after some seconds.

Yes, such a delay is quite understable but unfortunatelly my issue isn't about just a delay: it actually stays on the wrong language.

Samsung's Android interface was updated yesterday. After it, one of my phones returned to proper language in TopodroidX. The other remains.

In Android Studio the language setting stopped working for months now.

I managed to check that the code that saves the new language setting seems to be working fine even on Android Studio.

I am having trouble finding the code that uses the language setting to translate the strings. Can you point me where such a code is?

marcocorvi, 2024-04-09:
the language is managed by TDLocale

unlike other activites the setting activity does not have
onConfigurationChange
but there is a reload of the preferences.
i dont recall why.

is the language wrong only in the setting main page, or also in the
subpages ?

marcocorvi, 2024-04-09:
which android version do you have on your phones ?
what is the language setting in AndroidStudio ?
which android sdk are you using in your android studio ?

rsevero, 2024-04-09:
> which android version do you have on your phones ?

14 on both. And on Android Studio also. The update I mentioned before was on "One UI" which , is, AFAICT, a Samsung creation. It's Samsung Android interface.

> what is the language setting in AndroidStudio ?

It was english. When I changed to portuguese, the app started working in portuguese and I can go back and forth between portuguese and english inside the app. Other languages still don't work. I wonder if it has something to do with the info here: [https://developer.android.com/guide/topics/resources/multilingual-support#specify-the-languages-your-app-supports](https://developer.android.com/guide/topics/resources/multilingual-support#specify-the-languages-your-app-supports)

Unfortunatelly including 
`resConfigs 'en', 'pt', 'it', 'ru', 'de', 'cn', 'hu'`
in _build.gradle_ didn't solve the issue but I am quite sure that the issue is about some kind of optimization Android Studio is doing in the build process and not including language info for all available languages.

> which android sdk are you using in your android studio ?

Was already using 34. Target SDK in _build.gradle_ was 33, just changed to 34.

marcocorvi, 2024-04-16:
closed because issue resolution was not pursued
and the issue could be reproduced.

closed 2024-04-16 by marcocorvi