marcocorvi/topodroid #102: "Split sketch" not adding symbols in new sketch

rsevero, 2024-04-17:
**Describe the bug**
"Split sketch" not adding symbols in new sketch.

**To Reproduce**
Steps to reproduce the behavior:
1. Open to a sketch.
2. Click on "Rename/delete" in the "3 dots" menu 
3. Change the name of the sketch
4. Click on "Split"
5. Draw a line around some symbols in the sketch
6. See that the symbols disappear from the original sketch as intended
7. Open the new sketch
8. See that it is empty.

**Expected behavior**
The new sketch should have the symbols that were removed from the original sketch.

**Device:**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.61

**Additional context**
If I set "Copy items" in the "Rename/delete" dialog, the selected symbols are not deleted from the original sketch but they still don't appear in the new sketch.

rsevero, 2024-04-18:
Apparently the issue was with the debug version I installed from Android Studio. Sorry for the noise.

marcocorvi, 2024-04-18:
pleaae, solve all the issues that arise with the debug version you install
from android studio:
find the *cause* of the problem,
and submit a *patch* for the sources or for the android studio project.

rsevero, 2024-04-20:
> pleaae, solve all the issues that arise with the debug version you install from android studio: find the *cause* of the problem, and submit a *patch* for the sources or for the android studio project.

I will try but I have no experience with, Java, nor with Android development and neither with Android Studio so these strange errors are hard to track.

closed 2024-06-11 by marcocorvi