marcocorvi/topodroid #107: External "GNSS App" setting getting stuck in GPS Logger

rsevero, 2024-04-20:
**Describe the bug**
External "GNSS App" setting getting stuck in GPS Logger

**To Reproduce**
Steps to reproduce the behavior:
1. Install GPS Logger in your android device
2. Set "Survey data > Location > GNSS App" to GPS Logger
3. Try to import a geo point
4. See that GPS Logger is mentioned on the import page title
5. Set "Survey data > Location > GNSS App" to none
6. Try to import a geo point
7. See that GPS Logger is still mentioned on the import page title despite the "Survey data > Location > GNSS App" setting mentioning "none"

**Expected behavior**
The new "GNSS app" setting should be honored.

**Device (please complete the following information):**
 - Samsung Galaxy 23 FE and Motorola A32
 - Android 14
 - TopoDroid 6.2.61

marcocorvi, 2024-04-20:
as stared in the manusl, this setting is the "preferred" gnss app..

closed 2024-04-20 by marcocorvi

rsevero, 2024-04-20:
_page_fixed_import.htm_ does not mention any preferred app. Where can I find this info?

And what does being "preferred" means? Other options are silencly ignored?

marcocorvi, 2024-04-21:
> *page_fixed_import.htm* does not mention any preferred app. Where can I
> find this info?
settings

> And what does being "preferred" means? Other options are silencly ignored?
preferred = first choice

rsevero, 2024-04-21:
> On Sat, Apr 20, 2024, 22:53 Rodrigo Severo ***@***.***> wrote: *page_fixed_import.htm* does not mention any preferred app. Where can I find this info?
> settings
> And what does being "preferred" means? Other options are silencly ignored?
> preferred = first choice

I understand that GPS Logger might be the first choice. But why it is getting stuck in GPS Logger.

I change the setting of external app but when I try to acquire the coordinates of a new point, Topodroids interface is back at GPS Logger.

How can I make Topodroid honor a "external gnss app" setting that's not GPS Logger?

marcocorvi, 2024-04-21:
probably you have only gps logger, therefore there is no choice