marcocorvi/topodroid #113: Two long taps on shots in shot list doesn't select all shots in between if second long tap is above the first

rsevero, 2024-05-03:
**Describe the bug**
Two long taps on shots in shot list doesn't select all shots in between if second long tap is above the first

**To Reproduce**
Steps to reproduce the behavior:
1. Go to shots list window
2. Long tap on a shot
3. Long tap on another shot a few shots above the first
4. See that only the two taped shots are selected and not also all shots in between

**Expected behavior**
After the second long tap I expected that all shots in between the two taped shots were also selected as it happens when the second long tap is on a shot below the first.

**Device (please complete the following information):**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.64

marcocorvi, 2024-05-03:
this had already come up some time ago,
and it is the expected behaviour.

it is described in man page 7

rsevero, 2024-05-03:
> this had already come up some time ago, and it is the expected behaviour. it is described in man page 7

I just re-read man page 7 and tested current behaviour a little more. I believe it not quite intuitive as the second long tap always affects the items above it.

I wrote a different implementation where the second long tap always affects the items between itself and the previous taped item, above or below itself.

It seems more intuitive to me. It's on commit 4944b749e

Can you check if it seems as a good alternative to you?

marcocorvi, 2024-05-05:
it seems good.

the proposed solution requires to set/clear one end of the sequence and
set/clear the other end with a longtap.
a longtap set after a clear, or a longtap clear after a set, have the
effect of a short tap.

you should experiment with long survey listings and the multiselection in
different display mode of the survey data.

there is a minor problem: once one of the two lastpos is set, where are
they both cleared?

rsevero, 2024-05-05:
> there is a minor problem: once one of the two lastpos is set, where are they both cleared?

On the current code they are never both cleared but I don´t think that is a problem. Do you see some situation where that would be an issue?

marcocorvi, 2024-05-05:
> there is a minor problem: once one of the two lastpos is set, where are
> they both cleared?
>
> On the current code they are never both cleared but I don´t think that is
> a problem. Do you see some situation where that would be an issue?
when you reenter multiselection

marcocorvi, 2024-05-05:
the proposed update has been merged in the main

rsevero, 2024-05-05:
> On Sun, May 5, 2024, 15:06 Rodrigo Severo ***@***.***> wrote: there is a minor problem: once one of the two lastpos is set, where are they both cleared? On the current code they are never both [...]

I believe the _reentering multiselection_ issue is solved with commit e163784cd.

marcocorvi, 2024-05-06:
clear() removes the items from the adapter.
my concern was about having state variables not reset to default values at
the end.
although it did not matter (as it was)

i added a visual feedback to highlight the last chosen shot.

marcocorvi, 2024-05-06:
> Besides overriding *clear()* I had also included a reset of *lastPosAdd*
> and *lastPosRemove* in *clearMultiSelect()*. I saw you removed those too.
> Shouldn't those be there?

they are in the method that reset the multiselection state variables

rsevero, 2024-05-06:
> clear() removes the items from the adapter. my concern was about having state variables not reset to default values at the end. 

I overrod _clear()_ because it was the only "cleanup" code called at the end of "station swap" multiselection action.

The other, probably simplier action, would be to explicitly call _clearMultiSelect()_ at the end of _doSwapBlocksName()_ but I thought that, if the items are removed from the adapter, it would not be a bad idea to reset state varaibles.

Anyway, right now we are back at the situation where there is state left after swap stations. My last code had prevented that and prevented state leak on actions I tested.

marcocorvi, 2024-05-06:
you are right.
i restored the method clear()

many actions call updateDisplay which reload the list, which is heavy.

it would be nice to have more fine-grained calls, to the necessary actions.
for example swap could call a refresh only on the views of the swapped
blocks