marcocorvi/topodroid #116: page_sketch_mode.htm mentioned in code but never shown

rsevero, 2024-05-13:
**Describe the bug**
_page_sketch_mode.htm_ mentioned in code but never shown

**Expected behavior**
I tried to see _page_sketch_mode.htm_ as a help page in TopDroidX but only saw other similar pages like _page_drawing_refs.htm_.

I see _page_sketch_mode.htm_ is mentioned in code but couldn't actually open it in the app.

Is it used somehow? How can I access it in the app?

Or can it be removed?

**Device:**
 - Samsung Galaxy 23 FE
 - Android 14
 - TopoDroid 6.2.65

closed 2024-05-13 by marcocorvi

rsevero, 2024-05-14:
> if the reference is not reachable it can be moved in a "unused" space. otherwise it should be left in the sources.

AFAICT it isn´t used (i couldn´t find a way to open the dialog that uses it as help page) but I am not entirelly sure it isn't actually used.

It's called at line 2298 of _SketchWindow.java_ but I don't know how to get to it using the app.

Can you confirm is it really not used?

marcocorvi, 2024-05-14:
which tools did you use to analyse the code to tell it is not used ?

rsevero, 2024-05-14:
> which tools did you use to analyse the code to tell it is not used ?

I don´t understand your question.

The "tool" I used is Android Studio. With it I saw that _page_sketch_mode.htm_ is mentioned at line 2298 of _SketchWindow.java_ as I already mentioned before.

Using the app I can't find how to actually run the code at line 2298 of _SketchWindow.java_ which is why I think this help page might not be used anymore.

marcocorvi, 2024-05-14:
some static analyzer of the code.

with visual code inspection it is not easy to say if a code point is
unreachable.

using the app is good, but it can only prove that something is used, not
that something is not reachable.