marcocorvi/topodroid #119: Help button on "Projected profile azimuth" dialog does not work

rsevero, 2024-06-06:
**Describe the bug**
Help button on "Projected profile azimuth" dialog does not work

**To Reproduce**
Steps to reproduce the behavior:
1. Go to new sketch
2. Click on "projected profile" checkbox
3. Click on "Ok"
4. Click on "?"
5. See that the help page is not presented

**Expected behavior**
For the "page_projection.htm" help page to be presented

**Device:**
 - Samsung Galaxy 23FE
 - Android 14
 - TopoDroid 6.2.72

rsevero, 2024-06-06:
Fixed with commit 29f44d322

marcocorvi, 2024-06-06:
why not initLayout ?

rsevero, 2024-06-06:
> why not initLayout ?

I tried it but the layout of the dialog box changed: the title disappeared and the dialog box got larger.

And it was not used before, just "setContentView".

Appart from that, yes, "initLayout" would be better, for sure.

marcocorvi, 2024-06-06:
that is probably a problem of the layout file

rsevero, 2024-06-06:
> that is probably a problem of the layout file

Can you try to fix the layout so "initLayout" would work?

I don't know anything about layouts in java. It's the "ProjectionDialog" layout.

marcocorvi, 2024-06-06:
fixed.
please revert your changes.

rsevero, 2024-06-06:
Done

closed 2024-06-11 by marcocorvi