marcocorvi/topodroid #123: 35 sdk Topodroid dies when opening a survey with Android 14

rsevero, 2024-07-30:
**Describe the bug**
35 sdk Topodroid dies when opening a survey with Android 14

**To Reproduce**
Steps to reproduce the behavior:
1. Install any 35 SDK Topodroid version 
2. Try to open any existing survey or create a new survey
4. See Topodroid closing immediately

**Expected behavior**
Newer Topodroid should work on Android 14 with SDK 34. Up to version 6.2.75 everything is fine. 

**Device:**
 - Samsung 23FE
 - Android 14
 - TopoDroid 6.2.82

marcocorvi, 2024-07-30:
>    3. See Topodroid closing immediately

write permission ?

marcocorvi, 2024-07-30:
i cannot check android 14.
therefore the issue is assigned to R. Severo

rsevero, 2024-08-01:
I believe commit [b4e6c26](https://github.com/marcocorvi/topodroid/commit/b4e6c26c0228e928c08e0700efa8c41f22f0a143) fixed this issue.

I am not entirely sure as I can't create a proper update and try it on my phone.

For a completely new installation, survey opening and creation are working again.

marcocorvi, 2024-08-01:
indeed there was a typo in the table create.
it made the table create fails so the DB had no tables.

this solves the issue of topodroid not listing the paired devices
(they could not be inserted in the DB)

however, surveys are in a different DB.

marcocorvi, 2024-08-01:
oops, i was correcting a similar error in the device DB,
i realized you already corrected the one in the survey DB.

yes, now the issue is fixed

marcocorvi, 2024-08-01:
fixed and fixed also the failure to list distox

closed 2024-08-01 by marcocorvi

rsevero, 2024-08-02:
My local installation had no "reftype" columns despite being on DB version 54. I had to manually add them.

I believe commit 967e72ee2 will fix this automatically for other users that might be in the same situation I was.

marcocorvi, 2024-08-02:
i saw the commit.

> My local installation had no "reftype" columns despite being on DB version
> 54.
how did that happened ?

> I had to manually add them.
>
> I believe commit 967e72e
> will fix this automatically for other users that might be in the same
> situation I was.
how could that happen ?

rsevero, 2024-08-02:
I am not sure. I tried almost all version between 6.2.75 and 6.2.84. I started having problems with the first version you mentioned "SDK 35 test".

To make TopoDroid work again I had to go back to 6.2.75 until now when I manually added the "reftype" field in the 3 media tables to use my regular data. When I did a clean install everything was working just fine.

Unfortunatelly I didn't check each version to understand exactly how this could have happened.

commit https://github.com/marcocorvi/topodroid/commit/967e72ee2d95d1233f572454e7dc8a3a0942ada4 is not pretty but prevents users getting stuck at version 6.2.75 as I were.

Please fill free to undo it if it looks too ugly.

marcocorvi, 2024-08-03:
i get an error message about duplicate column name.
apparently it has no effect, but we will take care of that later.

rsevero, 2024-08-03:
> i get an error message about duplicate column name. apparently it has no effect, but we will take care of that later.

Fixed with commit 68df32996.

Sorry for the confusion.