marcocorvi/topodroid #124: Thanks!!

agseaton, 2024-09-01:
Not a bug, just a comment: thank you for your hard work!

I just updated to the latest version of topodroid after a few months and I'm really impressed with the progress that has been made since I last used it.

Here are just a few of the features that I noticed have been added, and will make a huge difference for me, and I'm sure to other users:
 - Being able to easily join lines in sketches (will save lots of time cleaning up scraps for therion)
 - Fixed Bezier curve functionality
 - Customisable colours for splays
 - Ability to show/hide splays on profile/plan
 - Ability to select a group of shots and do bulk actions, e.g. colour sets of splays, rename shots, etc. (this will also save huge amounts of time!)

I'm sure there are also plenty of other changes and fixes that I haven't noticed.

Of course, there's an endless list of things that could be improved, and it is important to point these out and work on them. It can be incredibly frustrating when the software doesn't do what you want!

But at the same time, we should also recognise the huge amount of work that has gone into this project, and how helpful Topodroid is for the community. I think there are many people who might not realise just how much time it takes to develop software like this.

So again, thanks, and I'm looking forward to seeing Topodroid get even better!

marcocorvi, 2024-10-31:
which of these feature requests is not already in the code ?

closed 2024-10-31 by marcocorvi

rsevero, 2024-10-31:
> which of these feature requests is not already in the code ?

I believe you didn't understand [marcocorvi](https://github.com/marcocorvi): the list of features [agseaton](https://github.com/agseaton) mentioned in his post is a list of features he is thankfull for already existing in the code. As am I.