marcocorvi/topodroid #125: "team" in exported Therion files are not separated by names

rsevero, 2024-10-27:
**Describe the bug**
"team" in exported Therion files are not separated by names

**To Reproduce**
Steps to reproduce the behavior:
1. Set more than one name in team separated by commas
2. Export centerlline data in Therion format
3. Open the exported th file
4. See that all names are in the same commented "team" line 

**Expected behavior**
One oncommented "team" line per person.

**Device:**
 - Samsung 23 FE
 - Android version 14
 - TopoDroid version 6.3.1

**Note**: this week I will be back home from my current expediotion. I will try to fix this.

marcocorvi, 2024-10-27:
This is NOT A BUG, it is TopoDroid EXPECTED behaviour.
TopoDroid has only a single string for the whole "team"

closed 2024-10-27 by marcocorvi

rsevero, 2024-10-27:
Ok. So I maybe I can update the manual13.htm page that says:

```
The team names are listed but the command is commented.<br>
The TopoDroid team string is split into the Therion <i>team</i> commands: team persons must be listed
with the lastname possibly preceded by the firstname initial(s). For example
the team string "A.W. Smith, B. Ford" becomes <i>team "A.W. Smith"</i> and <i>team "B. Ford"</i>.<br><p>
```

But the code to actually create one team line per member was already there in the code  tree. Just needs a little fix.

marcocorvi, 2024-10-27:
when did the code go in?
in any case that code does not seem foolproof.

properly separating names in the "team" string is a pain in the neck:
- names could be written without initials
- first names might be written completely
- one name might more than one first name
- a last name may be composed of two words
- one might add affiliations, possibly sorrounded by parenthesis, or square
brackets, or just plain.
- separators can vary: comma, semicolon, dash, tab, ....)

in other words there is too much choice in writing a free team string, to
expect that it is properly parsed.

one can
(1) either require a  (rather) strict format for the team string
(2) replace the input of the free-text string with a dialog that encodes
the team names, say, in xml or a custom structured string.

solution (1) is hopeless in my opinion.
people do not read user manuals and write whatever they believe is the
correct syntax,
and they expect it works too!

solution (2) is an extra burden for the user.
it may be reserved to therion users, that are accustomed to additional
extra complexity.

- team field should be editable for nornal users, but only clickable for
therion users.
- when therion user click on the team field the team editing dialog is
shown.
- the dialog has a couple of lines for team members and the possibility to
add additional lines (maybe with a max of 6, which is already three persons
that could make better use of their time).
the dialog format the team string with some special separator that is used
by the export dialog to parse it.
it is also used by the dialog when it is re-opened.

rsevero, 2024-10-27:
I was planning a simplier solution:

1. Replace semicolons and vertical bars for commas
2. Split team string on commas
3.  Trim each split-result
4. Replace repeated spaces with single space on each split-result
5. Count how many spaces are in each split-result: for 1 space do nothing; for more than one space, substitute the last space for a /; for zero spaces add the fixed suffix SURNAME
6. Put each slipt-result in a separated team line surrounded by double quotes.

It definitely isn't perfect but I believe it looks ways better than the current situation and is way nearer to the functionality described on the manual.

What do you think?

marcocorvi, 2024-10-27:
> I was planning a simplier solution:
>
>    1. Replace semicolons and vertical bars for commas
>
> this assumes that there are no other separator of team member.

what about people that do not use any separator ?

>
>    1. Split team string on commas
>    2. Trim each split-result
>    3. Replace repeated spaces with single space on each split-result
>    4. Count how many spaces are in each split-result: for 1 space do
>    nothing; for more than one space, substitute the last space for a /; for
>    zero spaces add the fixed suffix SURNAME
>
> why to add a suffix ?
a user could just list lastnames of team members

>
>    1. Put each slipt-result in a separated team line surrounded by double
>    quotes.
>
> It definitely isn't perfect but I believe it looks ways better than the
> current situation and is way nearer to the functionality described on the
> manual.
>
> What do you think?
if you really want separated team members, you have to make the user write
them separately.
then you can format them into a string with a separator that you control
(after polishing of extra white spaces and wierd chars)

rsevero, 2024-10-27:
> On Sun, Oct 27, 2024, 17:17 Rodrigo Severo ***@***.***> wrote: I was planning a simplier solution: 1. Replace semicolons and vertical bars for commas this assumes that there are no other separator of team member.
> what about people that do not use any separator ?

All the names will be treated as a single name: TopoDroid can't guess from thin air what people want.

> 1. Split team string on commas 2. Trim each split-result 3. Replace repeated spaces with single space on each split-result 4. Count how many spaces are in each split-result: for 1 space do nothing; [...]

Because Therion demands that there are 2 parts for the name of each person: either there are actually 2 names or the 2 parts are separated by a slash (/). If the is only one parts, I need to add a second one so Therion  won't complain.

> a user could just list lastnames of team members

Yes, that's right. I would only add a suffix for the slipt-results that have no spaces inside whatsoever, i.e., person names with only one part.

> 1. Put each slipt-result in a separated team line surrounded by double quotes. It definitely isn't perfect but I believe it looks ways better than the current situation and is way nearer to the [...]

I can include in the team editing dialogs a note to separate names with commas. If the user includes wierd chars, there they will remain as by user choice.

As I said, such an implementation won't by perfect but will be usable and easy for the ones interested. If you agree to it, obviously.

marcocorvi, 2024-10-27:
> On Sun, Oct 27, 2024, 17:17 Rodrigo Severo *@*.***> wrote: I was planning
> a simplier solution: 1. Replace semicolons and vertical bars for commas
> this assumes that there are no other separator of [...]
the dialog must take care of the separators, not the user

rsevero, 2024-10-27:
Sure, the dialog will take care of all the separators I can think about.

marcocorvi, 2024-10-27:
> On Sun, Oct 27, 2024, 17:17 Rodrigo Severo *@*.***> wrote: I was planning
> a simplier solution: 1. Replace semicolons and vertical bars for commas
> this assumes that there are no other separator of [...]

therion does not require two parts for the name of each person

rsevero, 2024-10-27:
From Therion manual:

```
• person . a person’s first name and surname separated by whitespace characters. Use
‘/’ to separate first name and surname if there are more names.
```

marcocorvi, 2024-10-27:
even if the manual says "and", therion does not complain if a person has
only one name, which can happen.

rsevero, 2024-10-27:
Great. If Therion does not complain about a person having only one name, there is no need for the suffix.