marcocorvi/topodroid #129: Triangle loop closure not working: inconsistent results

rsevero, 2024-11-27:
**Describe the bug**
Triangle loop closure not working: inconsistent results

**To Reproduce**
Steps to reproduce the behavior:
1. Set "Loop Closure" to "triangles"
2. Create an empty survey
3. Manually create 3 shots:
```
From To Len Az Clino
0    1  3   12 0
1    2  4   78 0
2    0  5   43 0
```
4. Create a new sketch
5. See that the 2-0 shot is way smaller than 5 meters.

**Expected behavior**
The length of the shots should not change (or at least not change significantly) when applying triangle loop closure, just the azimuths should be properly adjusted.

**Device:**
 - UleFoneArmour 24
 - Android 13
 - TopoDroid 6.3.6-35

marcocorvi, 2024-11-28:
The trilateration minimization fails if the starting point is close to a local minimum far from the global minumum.
This can be due to inconsistent data.
I added a toast message to warn the user about the failure.
It will be in v. 6.3.7.

rsevero, 2024-12-14:
Current Trilateration code not properly adjusting the folowing rectangular triangle:

```
From To Len  Az     Clino
0    1  1.38 137.3  -1.9
1    2  3.73 68.1    1.0
2    0  3.48 198.4  -1.1
```

AFAICT, it is calculating everything properly but when drawing the shots on the drawing window, the calculated adjustment is being ignored.

rsevero, 2024-12-14:
Correction: if you type the above survey data, trilateration works nicely but if you use the attached survey (which has the same data) it does not.
[deCur_2.zip](https://github.com/user-attachments/files/18136323/deCur_2.zip)

marcocorvi, 2024-12-14:
zip database too new: how can that be ?

rsevero, 2024-12-14:
> zip database too new: how can that be ?

Sorry. It's from the new code I am working on.

Here is a version you will be able to import:

[deCur_2.zip](https://github.com/user-attachments/files/18136975/deCur_2.zip)

marcocorvi, 2024-12-14:
the initial step is hardcoded to 10 cm.
that's too much for the example (in one case it already says failed).
with initial step adjusted to the legs length it says "failed trilateration" always.
this is an **improvement**, not a **fix**.

it is minimization process.
the result depends from the initial point and the rate of descent toward the minimum.
if the initial point is not near a local minimum or this has low barrier the point can go 
astray

rsevero, 2024-12-14:
But why does the manually added version gets properly adjusted and the automaticaly version does not?

The data is the same, the only difference I can see one from the other is that the downloaded version has duplicated shots.

And despite failing to properly adjust the automatic version believes it succeeded here with the message 

> Trilateration error 0.26

Doesn't this message means that trilateration believes it succeeded?

rsevero, 2024-12-14:
I believe the trilateration code is succeding but it's adjustment is getting lost _TDNum.java_ line 1130 on _makeShotFromTmp_ call.

marcocorvi, 2024-12-14:
i will check that

marcocorvi, 2024-12-14:
with delta = 10 cm the two triangles differ.
it can happen as i explained before, if the minimization surface is irregular
near the initial point.
trilateration can fail and it can stray to different points.

rsevero, 2024-12-14:
I believe I understand why it could fail but in this case, at least here, it is not failing.

The trilateration code is succeding but the calculated adjustment is getting lost.

Does the trilateration code fails there with you? With the _Trilateration failed_ error message from line 2150 of _TDNum.java_?

Here it succeeds with the _Trilateration error 0.26_ info message from line 2170 of _TDNum.java_.

marcocorvi, 2024-12-14:
where the calculated adjustment fails to be used ?

rsevero, 2024-12-14:
> where the calculated adjustment fails to be used ?

At the drawing window.

With the manually added shots survey I get the expected rectangular triangle I expected.

With the automatically downloaded shots survey, despite the shot data being the same as the manually shots, I get a triangle that is almost isosceles.

marcocorvi, 2024-12-14:
i think that tdnum should take into account also the error reported by the trilateration, to decide whether to accept it or not.

marcocorvi, 2024-12-14:
i meant:  in which part of the sources ?

marcocorvi, 2024-12-14:
todo: take better account of final error of minimization.
better estimate for others minimization parameters

rsevero, 2024-12-15:
> todo: take better account of final error of minimization. better estimate for others minimization parameters

With the new delta both surveys are failing trilateration: the one with manually entered data and the one with automatic downloaded data.

rsevero, 2024-12-15:
> i meant: in which part of the sources ?

I got back the old delta code to try to discover were the 2 surveys diverge. Still couldn't find the point. Will keep looking tomorow.

rsevero, 2024-12-15:
I believe the issue is related to the fact that at the manually entered data survey each shot has only one data block and on the automatically downloaded each shot has 2 data blocks (I reduced to 2 measurements per shot) but I can´t fiond in the code where this difference actually matters and results in such different results from both surveys.

marcocorvi, 2024-12-18:
to see what happens one should follow the path in configuration space described by the system during the minimization.
my guess is that the paths diverge pretfy soon, when the steps are big and they can jump across ridges of the minimization function.

marcocorvi, 2024-12-18:
i believe you are right.
there must be something wrong in the way the output of the trilateration is used.

rsevero, 2024-12-18:
Fixed with commit f5b26580e

marcocorvi, 2024-12-19:
i merged your fix with my changes.
left old cstr in NumShot commented.

tried to have a better decision for trilateration success/failure 
not sure that it is satisfactory: i reopen the issue.

need also clean-up of the logs

rsevero, 2024-12-19:
With my original simple tests the new code is failing all the time. The original code I started testing succeeded in 2 of my 3 simple cases. Apparently having near rectangular triangles is a big issue for current trilateration code.

marcocorvi, 2024-12-19:
indeed i added a test on angles

rsevero, 2024-12-20:
I see you created the _maxAngle()_ check. I don't think such a check should exist at all.

At least in the _Triangulation_ code I am finishing, I assume that the receive azimuths are arbitraty, complete garbage.

I don't know if such an assumption makes sense for _Trilateration_ as it works per aproximation but, if it can work without imposing limits on the change of azimuth, I believe it would be more usefull.

marcocorvi, 2024-12-20:
i have a problem with triangles: i dont know how to distinguish left- and right-handed.

rsevero, 2024-12-20:
You can't. At least not if you assume the received azimuths are arbitrary as I am doing.

In Triangulation code (available at the triangulation branch in github) I always assume the triangles are clockwise.

 I created an interface (and data table etc etc etc) so the user can, triangle by triangle, check, decide and eventually mirror the triangle. It's an option available at the station edit dialog in the sketch window. 

The whole process I imagined is described at page_triangulation.htm also available at the same branch. 

I believe trilateration code also needs this feature. It won't be hard to adapt it to work also in trilaterarion.

The main issue is that I implemented triangulation as a survey on-create-option. It can't be changed after survey creation. It is not a loop closure method.

But the reasons for that are others. We can talk about that latter if you prefer.

marcocorvi, 2024-12-22:
trilateration is an exploratory feature and i think it does not need to be completely worked out.
added some comments into the code to mark where the parameters could be.

closed 2024-12-22 by marcocorvi