marcocorvi/topodroid #141: Double counting cave length

RaymondMichael, 2025-07-06:
**Describe the bug**
TopoDroid is double counting the survey length. The reported length of the survey is twice what was entered.

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new cave project
2. Manually create a new fake shot. Create a shot that is 10 long in any direction. In the same dialogue, enter a back sight with the same length and the opposite azimuth.
3. Return to the main TopoDroid screen where you can select the different shots.
4. Long-hold on the cave to bring up the Survey Info screen
5. Tap the "i" with the circle around it
6. Look at the first "Legs" line.
7. See that the reported length is twice the length of the one fake shot that was created.

**Expected behavior**
The length of the cave should only count each shot once.

**Device (please complete the following information):**
 - Samsung Active Tab 3
 - 6.2.95-35

closed 2025-07-09 by marcocorvi

RaymondMichael, 2025-07-09:
What commit did the fix go in as? I don't see anything in the git log.

marcocorvi, 2025-07-23:
> I have been traveling, and I apologize for taking so long to get back to
> you.
> It seems like the front sight and the back sight are treated as separate
> legs. When you enter the front and back sight data at the same time in the
> same dialog box, would it be possible to have TopoDroid automatically flag
> the back sight data as not contributing to the legs' length?

this is possible for data entered manually.

with the latest version, topodroid reports the "survey length" which is the
leg length computed without counting siblings.

"backsight" would be a new flag, because it conveys different meaning than
other flags.

the only use i can forsee for a backsight flag is the computation of the
percentage of the survey that has been double-checked.
provided one takes at most only one check shot for each leg.

it might be useful if postprocessing programs (ie. therion) make some use
of that.
as far as i know a "backsight" flag is not supported in pc programs - but
my knowledge of pc programs is limited and i would be happy to get to know
which programs support it (and use it).

therefore a "backsight" flag would be useful only if available also for
distox shots.
having to mark each backsight distox by hand is not an option because it is
inconvenient.
i am not sure whether the flag can be set automatically, with the backsight
station policy.